Hexxeh/rpi-firmware #156: USB audio stuttering since update to 4.9 kernel (USB packet management regression?)

trulio, 2017-08-15:
Hi, 

this is an issue i already put on "https://github.com/raspberrypi/firmware" but as i see it doesn't seems to be addressed, i also report it here.

Here is the situation  : 
I'm using an synchronous USB DAC (USB adaptive mode).
see this product description if needed. (http://hifimediy.com/DACs/ready-made-dacs/Sabre-9018-DAC)

Before Kernel 4.9 (4.4 and others), il had a good experience with this dac since the usb driver improvements of years 2013-2014.

By good experience, i mean that when i launched a music, i had some stutering during the first seconds of playing after a system restaret but after that, everything was clean. 

But since upgrade to 4.9 kernel in february, this stutering issue occurs one, two, trhres times per hour during 10 to 20 seconds 

the description of stutering issue  is the following one : every second and during 10 to 20 seconds, you have some kind of glitch as if the usb sound card doesn't had nothing to play during half a second more or less.

Now i downgraded to last 4.4 kernel version of firmware to get rid of this issue but when i test the last release of the firmware, the issue is still present.

Regards,

Trulio

pelwell, 2017-08-15:
This is a duplicate - the relevant people (@P33M, really) have looked at and contributed to that issue, so creating a new issue doesn't help.

P33M currently has very little time to look at this problem, and USB is always going to be a best-effort service using this controller - it's amazing how well it performs as it is, and that's thanks to @ghollingworth and @P33M - but sit tight and we will get back to your issue eventually.

In the meantime you can help by gathering evidence of other people with the same problem - more data points - or find a use case that fails more rapidly.

popcornmix, 2017-08-15:
This is the wrong place to report firmware issues.
This repo is purely used for distribution of firmware through the rpi-update tool.

closed 2017-08-15 by popcornmix