Hexxeh/rpi-firmware #157: USB to serial Adapter not working since 4.9

ecdlguy, 2017-08-17:
Hi,

I'm using a USB to serial adapter with an pl2303 chip on a RPi2.

Using the 4.9 kernel, dmesg states:

`pl2303 ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32`

The adapter worked before without any problems.
Any ideas?

cheers,
Thorsten

ecdlguy, 2017-08-18:
Hi again,

please close this issue, it turned out it has nothing to do with the kernel.
There was another process accessing /dev/ttyUSB0 which caused the problems.

Sorry for the inconvenience.

cheers,
Thorsten

closed 2017-08-18 by popcornmix

pelwell, 2017-08-18:
Out of curiosity, and in case it affects somebody else, what was the process?

ecdlguy, 2017-08-23:
I've installed cuxd https://github.com/jens-maus/cuxd
which by default connects to all available serial ports
Unfortunately, I've installed this shortly before upgrading to the 4.9 kernel