Hexxeh/rpi-firmware #158: Hosed RP3 with Streach

Andyj9, 2017-09-01:
FYI -
Tried your firmware update - Raspberry PI 3 with the latest version of Raspbian (Stretch) installed.
Seem to be OK at first downloading and updating but then gave some errors about main not found, about 4 errors and quit. (Sorry - I closed the terminal session so do not have actual details)
Powered off my RP3 - and now, 100% NOTHING, I do have a power light in the front but the Ethernet never turns on and get zero display HDMI. It's dead...
Before all this I did see some lockups - may be I actually had a bad board to start with anyway.
Have you tested with the new Aug/2017 version of Raspbian?

Really hope I am the only one this happens to..

   Andy Johnson
   handyjohnson.com

pelwell, 2017-09-01:
I'm sorry you seem to be having problems - partial updates can be fatal. I've not heard of similar issues being reported, so it could be something about your SD card, power supply or Pi (hardware faults are _very_ rare, but not unheard of).

Your story makes me wonder how hard it would be to add a feature to rpi-update so it can run filesystem checks and try to install a self-consistent firmware and kernel on a card plugged into a Pi via a USB card reader. As it is now, you may be able to copy some user data off that way.

There should be no Raspbian/kernel dependencies severe enough to prevent a Pi from booting - the most I might expect is a missing feature or two. "Updating" to a firmware older than the first supporting that device can be fatal, at least on that device - anything will boot on an old enough Pi - but going forwards should always be safe. There are some hard kernel/firmware dependencies, although we try to avoid them, which is why rpi-update updates both at the same time.

Andyj9, 2017-09-01:
FYI - 
My RP3 is working now - the image was corrupted but when I re-imaged the SDCard everything started working again. When this happened, and from the screens (or lack of) that I was seeing - panic set in - Guess I have been spending to much time with my rooted android phones.. Thought a failed FW was on the RP board - finding out No not really, everything for Raspberry PI is on the  SDCard so replacing will solve 99.9% of issues.
Also - thank you for your fast response - hope this is interesting read for others as well.

   Andy