Hexxeh/rpi-firmware #159: Update from 4.9.35-v7 to 4.9.y - piface2 digital io no more detected

dupuisv, 2017-09-02:
after updating the raspberry pi3 to the latest firmware the piface2 board can no more be detected
- Jessie lite
- Piface 2
- Rapberry pi3 (using spi interface)
we did reproduce the same problem after a clean install of the latest jessie release.

pelwell, 2017-09-03:
1. Is this the PiFace Digital 2?
2. What is the commit hash or URL of the first firmware that doesn't work?
3. What is the simplest command to test it, and what output do you get on success and failure?

dupuisv, 2017-09-03:
Hello Phil,

 

1.       Yes. This one

https://www.dinoshop.ch/index.php?route=product/product <https://www.dinoshop.ch/index.php?route=product/product&path=1392_2420_1392_2420&product_id=125883> &path=1392_2420_1392_2420&product_id=125883

 

2.       Can you tell me how to install and test a particular firmware and to revert to the previous one.

So I can try.

                Probably a release between 1.5 and 31.8 . I do  not always do firmware update. Just for new Raspberry. 

3.

Python  (2.7.9)

import pifacedigitalio as piface

piface.init()

piface.digital_write(0, 1, 0)

 

 

Best regards

Reginald

pelwell, 2017-09-03:
Reginald,

You will find a list of the firmware releases on the [commits page](https://github.com/Hexxeh/rpi-firmware/commits/master). The long strings of characters on the right of each release are commit hashes - use the copy button to get the whole string, then on the Pi run `sudo rpi-update <hash>` to install a particular version. As I write this `sudo rpi-update` is equivalent to `sudo rpi-update 19a13325227cb59d44dbf6d235b5965439413aa4`.

Thanks for the test - can you say what output you get when it is successful and when it fails?

dupuisv, 2017-09-03:
Hello Phil,

 

I conduct a few test . Some kind of dichotomic search:

 

4.9.40              3202f1b16896029f9da1b074b0912177e8960b52     OK

4.9.41              b9becbbf3f48e39f719ca6785d23c53ee0cdbe49        OK

 

4.9.43              bf19fe7aa723ef8ddafdf8ca47a7205e35749ddc                  BAD

 

Traceback (most recent call last):

  File "DoorControl.py", line 1173, in <module>

    main()

  File "DoorControl.py", line 1018, in main

    init_door()

  File "DoorControl.py", line 558, in init_door

    piface.init()

  File "/usr/local/lib/python2.7/dist-packages/pifacedigitalio/core.py", line 166, in init

    raise failed_boards[0]

pifacedigitalio.core.NoPiFaceDigitalDetectedError: No PiFace Digital board detected (hardware_addr=0, bus=0, chip_select=0).

Python 2.7.9

 

So It’s start to fail with firmware 4.9.43

 

Regards

Reginald

pelwell, 2017-09-04:
That commit included a change to the maximum SPI bus speed (see https://github.com/raspberrypi/linux/issues/2165) because 500KHz was unnecessarily restrictive. Unfortunately the spidev API doesn't separate the concepts of the maximum speed and the current default speed, so it isn't possible to permit higher speeds without also changing the default.

As a result, unless clients are careful to either set a default speed (confusingly by using SPI_IOC_WR_MAX_SPEED_HZ) or to set the speed_hz property on each transfer then with the recent update they will find they are using a very high bus speed. This is going to be inconvenient for users of those client applications and libraries, but they've been lucky/unlucky so far that the default works for them. With the new maximum of 125MHz this will no longer be the case.

Reading the original issue again I see it is mentioned in a change to the `pifaceio` library (see https://github.com/bulletmark/pifaceio/commit/34ddceb72f2c8562d66b5f395227d9785221ff21). @XECDesign Can we update the Raspbian build to include that commit?

dupuisv, 2017-09-04:
Thank you for pointing the problem

 

The library  https://github.com/piface/pifacedigitalio  ,

That does support interrupts hasn’t been upgraded to set the speed explicitly.

 

I also think the piface library need sudo level to run.

https://github.com/bulletmark/pifaceio/commit/34ddceb72f2c8562d66b5f395227d9785221ff21

 

So we use the pifacedigitalio suggested by the piface2 maker.

 

Regards

Regi

Reading the original issue again I see it is mentioned in a change to the pifaceio library (see bulletmark/pifaceio@ <https://github.com/bulletmark/pifaceio/commit/34ddceb72f2c8562d66b5f395227d9785221ff21> 34ddceb). @XECDesign <https://github.com/xecdesign>  Can we update the Raspbian build to include that commit?

XECDesign, 2017-09-04:
The developer of the piface libraries no longer works for that company. We don't have Stretch packages for them.

dupuisv, 2017-09-04:
Thank for the information

I am making some change to pifacecommon (in the python source) to fix the problem.
Then I will look how to publish it.

Reginald

dupuisv, 2017-09-04:
Hello

The case can be closed. We update to 4.9.46 and apply few changes to set the SPI max rd/wr speed.

The update spi.py  is on  our local copy of pifacecommon-master.

If you can just tell me how to proceed to publish this update . That would be my first contribution

Regi

dBeltranGeze, 2017-10-27:
Is the fix published somewhere? I've tried to modify the speed in the spi_ioc_transfer struct on the pifacecommon-master/spi.py file but some of the commands fail.

dupuisv, 2017-10-28:
No. But send me your e-mail to info@dupuis-infornatique.ch and i can send you the modified python code. 
Reginald

dBeltranGeze, 2017-10-30:
Hi Reginald. I've tried to send you an email but my company's proxy raise an error. I'll share my email here: d.beltran@geze.com 

David

Ruffio, 2020-12-27:
@dupuisv is this issue still relevant? Else please close it, thanks :-)

closed 2020-12-27 by dupuisv