Hexxeh/rpi-firmware #161: SPI stopped working on Rpi B1 from 4.9.43+ onwards

scalvin1, 2017-11-07:
Hello,
SPI stopped working for me in the latest Raspbian stretch update. So I went back through the kernel versions here bisecting until I found the latest working version.
Latest working version for me is 4.9.41+, it does not work in 4.9.43+ anymore.
Is there a benign explanation? Did I miss a required configuration change?
Best regards,
Max

pelwell, 2017-11-07:
File this in https://github.com/raspberrypi/linux/issues and I'll be happy to help.

scalvin1, 2017-11-07:
I followed the instructions here years ago
https://www.raspberrypi-spy.co.uk/2014/08/enabling-the-spi-interface-on-the-raspberry-pi/
and also remember using this one
https://github.com/lthiery/SPI-Py.git
but I am not actually sure.
Which SPI library is recommended or up to date?

pelwell, 2017-11-07:
Some application must be failing, otherwise you wouldn't know that SPI is broken. Which application is it?

scalvin1, 2017-11-07:
pip install spidev -U
updated to latest version 3.2
Still not working.

It is a simple python script reading from mcp3008 A/D chip

import spidev
import time
import os

# Open SPI bus
spi = spidev.SpiDev()
spi.open(0,0)

# Function to read SPI data from MCP3008 chip
# Channel must be an integer 0-7
def ReadChannel(channel):
  adc = spi.xfer2([1,(8+channel)<<4,0])
  data = ((adc[1]&3) << 8) + adc[2]
  return data
....

pelwell, 2017-11-07:
Add:
```
spi.max_speed_hz = 500000
```
after `spi.open` to restore the old default/maximum speed.

scalvin1, 2017-11-07:
That fixed it. Thank you!

Ruffio, 2020-08-26:
This issue has been resolved and the GitHug user account has been deleted, therefore it says @ghost.
Please close this issue.