Hexxeh/rpi-firmware #164: hy28b overlay stops working 4.9.x+

CZyMfnIPC, 2018-01-15:
I am using the hy28b overlay to drive my LCD display, works fine in 4.1.x and 4.4.x, however when I update to any of the 4.9.x versions, it does not display on the screen.

Here is there relevant part of dmesg:
`
[    7.220507] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    7.251335] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned.
[    7.252110] pinctrl-bcm2835 20200000.gpio: /soc/gpio@7e200000/hy28b_pins: invalid brcm,pins value 1179673
[    7.252133] fb_ili9325: probe of spi0.0 failed with error -22
`

And from config.txt:
`
dtoverlay=hy28b,speed=48000000,fps=50,rotate=90,debug=0,resetgpio=25,ledgpio=18,txbuflen=327678
`

CZyMfnIPC, 2018-01-15:
Here is the dmesg from the working 4.4.50 firmware:
`
[    6.649209] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    6.659037] ads7846 spi0.1: touchscreen, irq 177
[    6.667779] input: ADS7846 Touchscreen as /devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1/input/input0
[    6.673143] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned.
[    6.682632] pinctrl-bcm2835 20200000.gpio: /soc/gpio@7e200000/hy28b_pins: invalid brcm,pins value 1179673
[    6.683294] fbtft_of_value: buswidth = 8
[    6.683366] fbtft_of_value: debug = 0
[    6.683379] fbtft_of_value: rotate = 90
[    6.683389] fbtft_of_value: fps = 50
[    6.683399] fbtft_of_value: startbyte = 112
[    7.330985] random: nonblocking pool is initialized
[    7.479648] graphics fb1: fb_ili9325 frame buffer, 320x240, 150 KiB video memory, 4 KiB DMA buffer memory, fps=50, spi0.0 at 48 MHz
[   11.583599] Adding 102396k swap on /var/swap.  Priority:-1 extents:4 across:184316k SSFS
[  446.405101] fb_ili9325 spi0.0: fbtft_update_display: start_line=239 is larger than end_line=0. Shouldn't happen, will do full display update

`

popcornmix, 2018-01-16:
Can you create this issue here: https://github.com/raspberrypi/linux/issues

closed 2018-01-16 by popcornmix