Hexxeh/rpi-firmware #166: Can't able to find out linux headers for 4.9.78-v7+

swanandb2, 2018-01-29:
pi@pilinux:~/socCan/can-isotp $ **sudo apt-cache search linux-headers**
aufs-dkms - DKMS files to build and install aufs
linux-headers-3.10-3-all - All header files for Linux 3.10 (meta-package)
linux-headers-3.10-3-all-armhf - All header files for Linux 3.10 (meta-package)
linux-headers-3.10-3-common - Common header files for Linux 3.10-3
linux-headers-3.10-3-rpi - Header files for Linux 3.10-3-rpi
linux-headers-3.16.0-4-all - All header files for Linux 3.16 (meta-package)
linux-headers-3.16.0-4-all-armhf - All header files for Linux 3.16 (meta-package)
linux-headers-3.16.0-4-common - Common header files for Linux 3.16.0-4
linux-headers-3.16.0-4-rpi - Header files for Linux 3.16.0-4-rpi
linux-headers-3.18.0-trunk-all - All header files for Linux 3.18 (meta-package)
linux-headers-3.18.0-trunk-all-armhf - All header files for Linux 3.18 (meta-package)
linux-headers-3.18.0-trunk-common - Common header files for Linux 3.18.0-trunk
linux-headers-3.18.0-trunk-rpi - Header files for Linux 3.18.0-trunk-rpi
linux-headers-3.18.0-trunk-rpi2 - Header files for Linux 3.18.0-trunk-rpi2
linux-headers-3.6-trunk-all - All header files for Linux 3.6 (meta-package)
linux-headers-3.6-trunk-all-armhf - All header files for Linux 3.6 (meta-package)
linux-headers-3.6-trunk-common - Common header files for Linux 3.6-trunk
linux-headers-3.6-trunk-rpi - Header files for Linux 3.6-trunk-rpi
linux-headers-4.4.0-1-all - All header files for Linux 4.4 (meta-package)
linux-headers-4.4.0-1-all-armhf - All header files for Linux 4.4 (meta-package)
linux-headers-4.4.0-1-common - Common header files for Linux 4.4.0-1
linux-headers-4.4.0-1-rpi - Header files for Linux 4.4.0-1-rpi
linux-headers-4.4.0-1-rpi2 - Header files for Linux 4.4.0-1-rpi2
linux-headers-4.9.0-4-all - All header files for Linux 4.9 (meta-package)
linux-headers-4.9.0-4-all-armhf - All header files for Linux 4.9 (meta-package)
linux-headers-4.9.0-4-common - Common header files for Linux 4.9.0-4
linux-headers-4.9.0-4-common-rt - Common header files for Linux 4.9.0-4-rt
linux-headers-4.9.0-4-rpi - Header files for Linux 4.9.0-4-rpi
linux-headers-4.9.0-4-rpi2 - Header files for Linux 4.9.0-4-rpi2
linux-headers-rpi - Header files for Linux rpi configuration (meta-package)
linux-headers-rpi-rpfv - This metapackage will pull in the headers for the raspbian kernel for the
linux-headers-rpi2 - Header files for Linux rpi2 configuration (meta-package)
linux-headers-rpi2-rpfv - This metapackage will pull in the headers for the raspbian kernel for the
linux-libc-dev-alpha-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-arm64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-armel-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-armhf-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-hppa-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-m68k-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mips64el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-mipsel-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-powerpc-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-powerpcspe-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-ppc64-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-ppc64el-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-s390x-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-sh4-cross - Linux Kernel Headers for development (for cross-compiling)
linux-libc-dev-sparc64-cross - Linux Kernel Headers for development (for cross-compiling)
raspberrypi-kernel-headers - Header files for the Raspberry Pi Linux kernel
linux-headers-4.9.68-v7+ - Linux kernel headers for 4.9.68-v7+ on armhf

I need those headers to work with socketCAN can you please help me with this?

DougieLawson, 2018-12-03:
There is a sane method. https://github.com/notro/rpi-source/wiki

swanandb2, 2019-08-13:
I did tested with sudo apt-get install build-essential linux-headers command and it did work but now 
for make command it throws following error ( build directory is missing)
pi@raspberrypi:~/uds/can-isotp $ sudo make
make -C /lib/modules/4.19.57-v7+/build M=/home/pi/uds/can-isotp modules
**make[1]: Entering directory '/lib/modules/4.19.57-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.19.57-v7+/build'
make: *** [Makefile:21: modules] Error 2**

Following are content of /lib/modules/4.19.57-v7+/ directory

pi@raspberrypi:/lib/modules/4.19.57-v7+ $ ls -l
total 2092
drwxr-xr-x 11 root root   4096 Jul 10 05:37 kernel
-rw-r--r--  1 root root 525300 Jul  8 18:32 modules.alias
-rw-r--r--  1 root root 546338 Jul  8 18:32 modules.alias.bin
-rw-r--r--  1 root root  11287 Jul  8 18:33 modules.builtin
-rw-r--r--  1 root root  12386 Jul  8 18:33 modules.builtin.bin
-rw-r--r--  1 root root 180509 Jul  8 18:32 modules.dep
-rw-r--r--  1 root root 250914 Jul  8 18:32 modules.dep.bin
-rw-r--r--  1 root root    302 Jul  8 18:33 modules.devname
-rw-r--r--  1 root root  61698 Jul  8 18:32 modules.order
-rw-r--r--  1 root root    352 Jul  8 18:33 modules.softdep
-rw-r--r--  1 root root 231312 Jul  8 18:32 modules.symbols
-rw-r--r--  1 root root 284469 Jul  8 18:32 modules.symbols.bin

Can you please help me for direction? Let me know if you need any other details.
pi@raspberrypi:/lib/modules/4.19.57-v7+ $ uname -a
Linux raspberrypi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

Ruffio, 2020-08-26:
@swanandb2 can this be closed?

swanandb2, 2020-08-26:
Yes please close it.

Ruffio, 2020-08-26:
@swanandb2 I can't close it, but you can...

closed 2020-09-09 by swanandb2