Hexxeh/rpi-firmware #167: Kernel Panic 4.9.79

crisconru, 2018-02-04:
Hi,
I use a distro call **KivyPie**, a **Raspbian based distro**. When I **upgrade** to the **kernel** version **4.9.79** and then I reboot, I usually **get a kernel panic**. Sometimes when I login or when I do an `apt update / upgrage`.

I didn't check in Raspbian, but I think the effect will be the same.

Thanks in advance

popcornmix, 2018-02-04:
Linux kernel issues are handled here: https://github.com/raspberrypi/linux/issues
You'll need to provide details on what the panic was to get any useful help.
Serial console output would be ideal, but a photo of screen would be the minimum.
Please confirm the issue occurs on raspbian.

closed 2018-02-04 by popcornmix

crisconru, 2018-02-04:
Sorry, then I discover that the kernel panic it is caused by the upgrade of the packets in the distro. And sorry to for my mistake (I didn't attach a screenshot or more information), I am not have experience with this.

Thanks for your attention ;-)