Hexxeh/rpi-firmware #168: Boot failure updating from 9fdd5ca79b8fea585ad92b36190158cca2b07eb6 to next versions

Sharktear, 2018-02-05:
Hi all,
I updated my Raspberry 3 firmware after about a month from last update and I get the last release (eb13a28aaa734d7ed75023cb29cd1a3ef4a7ef9c) but after reboot I can't connect to my Raspi.
So I tried many times to reboot without success.
The strange thing:
if I use a poor power supply the red led blynk but the Raspberry boot correctly... if I use a good 2,5A power supply the red led is stable on but the boot fail...
I tried to restore to the SD memory a previous backup (containing the 9fdd5ca79b8fea585ad92b36190158cca2b07eb6  release of the firmware) and all is fine and the raspberry boot correctly... then I tried to "rpi-update" againg and I fall in the same issue.
I tried to update to a different revision (c66bc1e2cc7f9fb2f23d774ee3fc1fdb21b35792) but same error...
So I think the problem is in a release next eb13a28aaa734d7ed75023cb29cd1a3ef4a7ef9c, but I don't know which...
Does anyone have a suggestion?

pelwell, 2018-02-05:
Something in firmware release https://github.com/Hexxeh/rpi-firmware/commit/23a007716a0c6a8677097be859cf7063ae093d27 ("kernel: Bump to 4.9.78") is causing problems for a few users that sound like what you are experiencing, but it isn't something we've been able to reproduce sufficiently to be able to pinpoint the issue. That undervoltage seems to make the problem go away is a useful datapoint.

One thing you could try is preventing the ARM speed from changing with the good power supply - add either `arm_freq=600` or `force_turbo=1` to config.txt. The first of these effectively disables turbo mode, but you will notice a significant slowdown. The second locks you permanently in turbo mode, but this will cause your Pi3 to heat up and has the potential to shorten the life of your Pi a little.

If you find that these settings make your Pi boot reliably then you could try a compromise setting of:
```
arm_freq=1000
force_turbo=1
```
Either way, let me know what you find.

pelwell, 2018-02-05:
This report is essentially a duplicate of https://github.com/Hexxeh/rpi-firmware/issues/165 - please see the updated firmware files linked from that issue and continue the discussion there.

closed 2018-03-06 by popcornmix