Hexxeh/rpi-firmware #169: git_hash from branch next commits cannot checkout kernel source from kernel clone

MrEngman, 2018-02-13:
Hexxeh/rpi-firmware, raspberrypi/firmware, raspberrypi/linux git repos broken again.

I use a script to automatically compile wifi drivers. The script uses the Hexxeh/rpi-firmware commit ID to select the kernel version to compile the drivers for. The script gets the git_hash value from the relevant Hexxeh/rpi-firmware commit to select and checkout the kernel version source from the kernel source clone I have to use for compiling the driver for the required kernel version. 

I have just tried to recompile some of the drivers I've compiled for kernel 4.14 to fix a bug in the driver but the script now says that the it `Cannot access Kernel Commit ID`.

It appears the value of git_hash in the commits in Hexxeh/rpi-firmware/next and raspberrypi/firmware/next are no longer valid and can not now be used to select and checkout the kernel version source the firmware gits originally pointed to. What has happened? It appears that the branch raspberrypi/linux/rpi-4.14.y may have been completely rewritten for some unknown reason. Why? If so why have the firmware gits not been updated?

As an example here is some output from my script.
```
pi@Pi-3-src:~ $ ./wifi-compiler4c.sh mt7612 2e88779c059bb4a866cfc63cf18feaf8fc2b78b8
Compiling mt7612 driver for Hexxeh rpi-firmware commit ID 2e88779c059bb4a866cfc63cf18feaf8fc2b78b8.
kernel = 4.14.4+
kernel7 = 4.14.4-v7+
build = #1060
build7 = #1060
kernel commit-id = d47bf375c59d4906fcaf3ad2c51cabb52c22e29e
git checkout -f d47bf375c59d4906fcaf3ad2c51cabb52c22e29e
Cannot access Kernel Commit ID.
pi@Pi-3-src:~ $ 
```
I receive similar output for kernel versions 4.14.0 to 4.14.17 . As you caused the problem can you supply a list of them as I don't want to have to search through pages and pages of commits in raspberrypi/Linux to find them.

pelwell, 2018-02-13:
When bringing up a new kernel there is often quite a lot of temporary patches followed by some tidying up and commit squashing. It is our standard policy to rebase new branches up until the point they become the mainstream (or are dropped for a later release).

4.14.18 is the first mainstream release from the rpi-4.14.y branch. There should be no further rebasing, but that cannot be guaranteed.

MrEngman, 2018-02-14:
Perhaps you should review this https://help.github.com/articles/about-git-rebase/ - especially the warning message.

and looking at some other git manuals it also says using rebase is not good if the repo is referenced from other places - e.g Hexxex/rpi-firmware and raspberrypi/firmware. I do often and this problem is a real pain and I'm sure it shouldn't be if handled correctly.