Hexxeh/rpi-firmware #172: Latest firmware forces sdram_freq to always be 400mhz

MaxOfS2D, 2018-03-04:
Hello,

I've been trying to overclock a Pi 3, and, to my surprise, vcgencmd always reports sdram_freq as 400. This was very intriguing since the default is supposed to be 450. (I was shooting for 600 with the help of overvoltage and schmoo parameters, and those were indeed taken into account).

I downgraded to the "default" kernel version with the help of this command I found in a closed issue:

`sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel`

And this solved the issue. sdram_freq was reported by vcgencmd to be 450 by default, and whatever number I set it to in /boot/config.txt if present.

popcornmix, 2018-03-06:
Yes, it's a side effect of the improved analogue audio that we haven't resolved yet.
See: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=195178

(changing sdram frequency dynamically can cause an audio interruption)

For now, add ``audio_pwm_mode=0`` to config.txt and your sdram frequency will return to the higher level.
We are working on a solution that only affects the sdram frequency when analogue audio is active, but that is not ready yet.

closed 2018-03-06 by MaxOfS2D

popcornmix, 2018-03-08:
This isssue should be resolved with latest rpi-update firmware.