Hexxeh/rpi-firmware #173: ft232 got slower

Osolemio, 2018-03-08:
Had a jessie release with 4.4.38-v7+ on Rpi3
All RS232-USB adapters worked like a charm: CP2102, PL2303, FT232
After upgrade to 4.9 and higher up to latest one: 4.14.24  cp2102 and pl2303 continue working fine but ft232. It started to make a delay about 7 sec. I'm not sure whether it is caused by ftdi kernel module or firmware.
Downgrade to 4.4.38 restores the normal operation.

Thank you!
Best regards.

pelwell, 2018-03-09:
Please describe the delay a bit more:
* When does it occur?
* How repeatable is it?

The differences in the driver between the 4.4 and 4.9 kernels are actually very small:
```diff
diff --git a/drivers/usb/serial/ftdi_sio.c b/drivers/usb/serial/ftdi_sio.c
index d3d6ec4..0c743e4 100644
--- a/drivers/usb/serial/ftdi_sio.c
+++ b/drivers/usb/serial/ftdi_sio.c
@@ -93,27 +93,27 @@ static int   ftdi_8u2232c_probe(struct usb_serial *serial);
 static void  ftdi_USB_UIRT_setup(struct ftdi_private *priv);
 static void  ftdi_HE_TIRA1_setup(struct ftdi_private *priv);
 
-static struct ftdi_sio_quirk ftdi_jtag_quirk = {
+static const struct ftdi_sio_quirk ftdi_jtag_quirk = {
        .probe  = ftdi_jtag_probe,
 };
 
-static struct ftdi_sio_quirk ftdi_NDI_device_quirk = {
+static const struct ftdi_sio_quirk ftdi_NDI_device_quirk = {
        .probe  = ftdi_NDI_device_setup,
 };
 
-static struct ftdi_sio_quirk ftdi_USB_UIRT_quirk = {
+static const struct ftdi_sio_quirk ftdi_USB_UIRT_quirk = {
        .port_probe = ftdi_USB_UIRT_setup,
 };
 
-static struct ftdi_sio_quirk ftdi_HE_TIRA1_quirk = {
+static const struct ftdi_sio_quirk ftdi_HE_TIRA1_quirk = {
        .port_probe = ftdi_HE_TIRA1_setup,
 };
 
-static struct ftdi_sio_quirk ftdi_stmclite_quirk = {
+static const struct ftdi_sio_quirk ftdi_stmclite_quirk = {
        .probe  = ftdi_stmclite_probe,
 };
 
-static struct ftdi_sio_quirk ftdi_8u2232c_quirk = {
+static const struct ftdi_sio_quirk ftdi_8u2232c_quirk = {
        .probe  = ftdi_8u2232c_probe,
 };
 
@@ -809,10 +809,10 @@ static const struct usb_device_id id_table_combined[] = {
        { USB_DEVICE(FTDI_VID, FTDI_PROPOX_ISPCABLEIII_PID) },
        { USB_DEVICE(FTDI_VID, CYBER_CORTEX_AV_PID),
                .driver_info = (kernel_ulong_t)&ftdi_jtag_quirk },
-       { USB_DEVICE(OLIMEX_VID, OLIMEX_ARM_USB_OCD_PID),
-               .driver_info = (kernel_ulong_t)&ftdi_jtag_quirk },
-       { USB_DEVICE(OLIMEX_VID, OLIMEX_ARM_USB_OCD_H_PID),
-               .driver_info = (kernel_ulong_t)&ftdi_jtag_quirk },
+       { USB_DEVICE_INTERFACE_NUMBER(OLIMEX_VID, OLIMEX_ARM_USB_OCD_PID, 1) },
+       { USB_DEVICE_INTERFACE_NUMBER(OLIMEX_VID, OLIMEX_ARM_USB_OCD_H_PID, 1) },
+       { USB_DEVICE_INTERFACE_NUMBER(OLIMEX_VID, OLIMEX_ARM_USB_TINY_PID, 1) },
+       { USB_DEVICE_INTERFACE_NUMBER(OLIMEX_VID, OLIMEX_ARM_USB_TINY_H_PID, 1) },
        { USB_DEVICE(FIC_VID, FIC_NEO1973_DEBUG_PID),
                .driver_info = (kernel_ulong_t)&ftdi_jtag_quirk },
        { USB_DEVICE(FTDI_VID, FTDI_OOCDLINK_PID),
@@ -873,6 +873,7 @@ static const struct usb_device_id id_table_combined[] = {
        { USB_DEVICE_AND_INTERFACE_INFO(MICROCHIP_VID, MICROCHIP_USB_BOARD_PID,
                                        USB_CLASS_VENDOR_SPEC,
                                        USB_SUBCLASS_VENDOR_SPEC, 0x00) },
+       { USB_DEVICE_INTERFACE_NUMBER(ACTEL_VID, MICROSEMI_ARROW_SF2PLUS_BOARD_PID, 2) },
        { USB_DEVICE(JETI_VID, JETI_SPC1201_PID) },
        { USB_DEVICE(MARVELL_VID, MARVELL_SHEEVAPLUG_PID),
                .driver_info = (kernel_ulong_t)&ftdi_jtag_quirk },
@@ -1014,6 +1015,9 @@ static const struct usb_device_id id_table_combined[] = {
        { USB_DEVICE(WICED_VID, WICED_USB20706V2_PID) },
        { USB_DEVICE(TI_VID, TI_CC3200_LAUNCHPAD_PID),
[... 106 lines cut ...]
-       else
-               priv->transmit_empty = 0;
-
-       len -= 2;
-       if (!len)
-               return 0;       /* status only */
        port->icount.rx += len;
        ch = packet + 2;
 
@@ -2433,8 +2443,12 @@ static int ftdi_get_modem_status(struct usb_serial_port *port,
                        FTDI_SIO_GET_MODEM_STATUS_REQUEST_TYPE,
                        0, priv->interface,
                        buf, len, WDR_TIMEOUT);
-       if (ret < 0) {
+
+       /* NOTE: We allow short responses and handle that below. */
+       if (ret < 1) {
                dev_err(&port->dev, "failed to get modem status: %d\n", ret);
+               if (ret >= 0)
+                       ret = -EIO;
                ret = usb_translate_errors(ret);
                goto out;
        }
```
We haven't made any downstream changes in the driver, so either it's a generic upstream problem or it is caused elsewhere - the USB driver, for example.

pelwell, 2018-03-09:
And this should be opened in the [Issues section of the raspberrypi/linux repo](https://github.com/raspberrypi/linux/issues).

Osolemio, 2018-03-09:
Thank you. Should I reopen this issue over there then?
Short description:
Suppose I have 4.4.38. Use a row communication via ftdi:
speed 19200 baud; line = 0;
min = 0; time = 1;
-brkint -icrnl -imaxbel
-opost -onlcr
-isig -icanon -iexten -echo -echoe -echok -echoctl -echoke

Protocol is very simle: I m'sending from the emulator byte by byte to RasPi. RasPi sends an echo for every byte respectively. So, I see on my emulator the last sent byte.
One session around 1kB takes about 400ms.

Then I upgrade the system via rpi-update, say, to 4.14
One session takes about 7 sec via ft232. For all other adapters it stays the same.

Then, I downgrade the system via rpi-update back to 4.4.38
the normal timing recovers

Thank you for the diff
Maybe I have to try compiling the old ftdi_sio.ko for the new kernel to find out whether the problem is in that.

pelwell, 2018-03-09:
>  Should I reopen this issue over there then?

It would be preferred.

> One session takes about 7 sec via ft232. For all other adapters it stays the same.
When you re-open, please a bit about the timing of the data that does arrive:
* We know the baud rate must be correct otherwise nothing would work (unless there is some auto-detection going on). How many pauses are there in the data flow and where are they?
* Are the pauses periodic? If so, how much data is sent between pauses and how long are the pauses?
* Can you tell if the data is delayed being sent, received or both?
* If you double the amount of data does the time increase to 14 seconds or stay roughly the same?

Osolemio, 2018-03-09:
the baudrate is correct. I'm sure :)
To measure the pauses I have to refine my receiving algorithm and add some additional logs. The problem is that any log entry has an influence on the communication timing.
Ok, I will make some additional testings, and open an issue in above mentioned thread.

Osolemio, 2018-03-14:
Hello, Phil

I have made some tests. The issue was in LOW_LATENCY fix commit:
https://www.systutorials.com/linux-kernels/94572/usb-serial-ftdi_sio-fix-extreme-low-latency-setting-linux-4-9-13/
A line: priv->flags = ASYNC_LOW_LATENCY;
I have to add now a new setting of this flag (in binary).
As for example, from CLI:
For pl2303 it is not supported:
/home/pi# setserial /dev/ttyUSB0 low_latency
Cannot set serial info: Inappropriate ioctl for device
But for ft232 it is.
/home/pi # setserial /dev/ttyUSB0 low_latency
/home/pi#
With this flag I could obtain the appropriate latency for ftdi chip.

So, we may close this issue.
Thank you for your time

closed 2018-03-14 by Osolemio