Hexxeh/rpi-firmware #179: Would appreciate dual boot

Osolemio, 2018-06-20:
Hello, 

As we know, PasPi can boot and continue on or form microSD or from USB stick either. That's hardcoded in /boot/config.txt
I used this feature for developed systems on RasPi, RasPi2 & RasPi3. But some time ago I had gone to RasPi based systems with microSD cards only and RO linux.
The question:
May we have an option, that allows us to start from microSD, continue from the USB stick when it's available, otherwise falls down back to microSD rootfs?
Why? I have a line of embedded devices based on RasPi. Some more or less small updates I can distribute via simple updates downloadable via web-interface . But sometimes I would need a complete microSD re-burning ant this is an issue: for big system updates, FW updates, recover end so on. Customers have no physical access to sd card, but USB slots. I spent many time for different attempts of re-flashing microSD "on the fly", but either one wasn't fully acceptable and safe. So, finally, I decided to leave my request here, because can't see another possibility beside realizing this feature: we start booting from SD, continue on from USB, then flash SD. If we had such an option I believe it would be fine. Even though a customer has to watch the system messages on console, but it's almost safe, because 1st partition will be burned very fast and RasPi will stay bootable even with other partitions corrupted. Thank you very much.

pelwell, 2018-06-20:
Please report this in the [raspberrypi/firmware](https://github.com/raspberrypi/firmware/) repo - these issues are for problems specific to this method of updating.

Osolemio, 2018-06-20:
Ok, I will. Thank you

closed 2018-06-20 by Osolemio