Hexxeh/rpi-firmware #180: uvcvideo: Failed to resubmit video URB (-1).

dssmex, 2018-06-29:
I have an issue when connect and open USB WebCam 

Kernel version:
Linux PM0000 4.14.52-v7+ #1123 SMP Wed Jun 27 17:35:49 BST 2018 armv7l armv7l armv7l GNU/Linux

comand: dmesg | grep uvcvideo
```
[    3.255711] uvcvideo: Found UVC 1.00 device USB Microscope (0c45:6353)
[    3.264222] uvcvideo 1-1.1.3:1.0: Entity type for entity Extension 4 was not initialized!
[    3.264235] uvcvideo 1-1.1.3:1.0: Entity type for entity Processing 3 was not initialized!
[    3.264240] uvcvideo 1-1.1.3:1.0: Entity type for entity Camera 1 was not initialized!
[    3.265618] usbcore: registered new interface driver uvcvideo
[  488.899030] uvcvideo: Failed to resubmit video URB (-1).
```

command: lsusb
```
Bus 001 Device 006: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 004: ID 2341:003d Arduino SA 
Bus 001 Device 005: ID 0c45:6353 Microdia 
Bus 001 Device 008: ID 0c2e:0ca1 Metrologic Instruments 
Bus 001 Device 007: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

command: journalctl -xar
```
jun 29 09:13:34 PM0000 wpa_supplicant[774]: wlan0: WPA: Group rekeying completed with 60:45:cb:cb:8f:18 [GTK=CCMP]
jun 29 09:12:35 PM0000 com.canonical.indicator.application[1068]: (process:1850): indicator-application-service-WARNING **: Application already exists, re-requesting properties.
jun 29 09:09:01 PM0000 CRON[2127]: pam_unix(cron:session): session closed for user root
jun 29 09:09:01 PM0000 CRON[2131]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && /usr/lib/php/sessionclean)
jun 29 09:09:01 PM0000 CRON[2127]: pam_unix(cron:session): session opened for user root by (uid=0)
jun 29 09:07:12 PM0000 kernel: uvcvideo: Failed to resubmit video URB (-1).
jun 29 09:02:52 PM0000 com.canonical.indicator.application[1068]: (process:1850): indicator-application-service-WARNING **: Application already exists, re-requesting properties.
jun 29 09:00:52 PM0000 systemd-logind[647]: New session c2 of user mpnuser.
-- Subject: A new session c2 has been created for user mpnuser
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- Documentation: http://www.freedesktop.org/wiki/Software/systemd/multiseat
-- 
-- A new session with the ID c2 has been created for the user mpnuser.
-- 
-- The leading process of the session is 1926.
jun 29 09:00:52 PM0000 systemd[1]: Started Session c2 of user mpnuser.
-- Subject: Unit session-c2.scope has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit session-c2.scope has finished starting up.
-- 
-- The start-up result is done.
jun 29 09:00:52 PM0000 sshd[1926]: pam_unix(sshd:session): session opened for user mpnuser by (uid=0)
jun 29 09:00:52 PM0000 sshd[1926]: Accepted password for mpnuser from 192.168.2.185 port 42754 ssh2
jun 29 09:00:49 PM0000 sshd[1926]: Failed password for mpnuser from 192.168.2.185 port 42754 ssh2
jun 29 09:00:47 PM0000 sshd[1926]: pam_unix(sshd:auth): authentication failure; logname= uid=0 euid=0 tty=ssh ruser= rhost=192.168.2.185  user=mpnuser
jun 29 09:00:07 PM0000 systemd[1]: Started Stop ureadahead data collection.
-- Subject: Unit ureadahead-stop.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ureadahead-stop.service has finished starting up.
-- 
-- The start-up result is done.
jun 29 09:00:07 PM0000 systemd[1]: Stopped Read required files in advance.
-- Subject: Unit ureadahead.service has finished shutting down
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ureadahead.service has finished shutting down.
jun 29 09:00:07 PM0000 systemd[1]: Starting Stop ureadahead data collection...
-- Subject: Unit ureadahead-stop.service has begun start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit ureadahead-stop.service has begun starting up.
jun 29 08:59:59 PM0000 blueman-mechanism[1882]: Exiting
jun 29 08:59:43 PM0000 com.canonical.indicator.application[1068]: (process:1850): indicator-application-service-WARNING **: Application already exists, re-requesting properties.
jun 29 08:59:35 PM0000 pkexec[1901]: mpnuser: Executing command [USER=root] [TTY=unknown] [CWD=/home/mpnuser] [COMMAND=/usr/lib/update-notifier/package-system-locked]
jun 29 08:59:35 PM0000 pkexec[1901]: pam_systemd(polkit-1:session): Cannot create session: Already running in a session
jun 29 08:59:35 PM0000 pkexec[1901]: pam_unix(polkit-1:session): session opened for user root by (uid=1000)
jun 29 08:59:29 PM0000 blueman-mechanism[1882]: loading Network
jun 29 08:59:29 PM0000 blueman-mechanism[1882]: loading Ppp
jun 29 08:59:29 PM0000 blueman-mechanism[1882]: loading RfKill
jun 29 08:59:29 PM0000 blueman-mechanism[1882]: loading Rfcomm
jun 29 08:59:29 PM0000 org.blueman.Mechanism[606]: (blueman-mechanism:1882): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
jun 29 08:59:29 PM0000 dbus[606]: [system] Successfully activated service 'org.blueman.Mechanism'
jun 29 08:59:29 PM0000 blueman-mechanism[1882]: Starting blueman-mechanism

```

comand: v4l2-ctl --all
```
Driver Info (not using libv4l2):
	Driver name   : uvcvideo
	Card type     : USB Microscope: USB Microscope
	Bus info      : usb-3f980000.usb-1.1.3
	Driver version: 4.14.52
	Capabilities  : 0x84200001
		Video Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 800/600
	Pixel Format      : 'YUYV'
	Field             : None
	Bytes per Line    : 1600
	Size Image        : 960000
	Colorspace        : Default
	Transfer Function : Default
	YCbCr Encoding    : Default
	Quantization      : Default
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 800, Height 600
	Default     : Left 0, Top 0, Width 800, Height 600
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 800, Height 600
Selection: crop_bounds, Left 0, Top 0, Width 800, Height 600
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 20.000 (20/1)
	Read buffers     : 0
                     brightness (int)    : min=-64 max=64 step=1 default=-8193 value=0
                       contrast (int)    : min=0 max=64 step=1 default=57343 value=36
                     saturation (int)    : min=0 max=128 step=1 default=57343 value=64
                            hue (int)    : min=-40 max=40 step=1 default=-8193 value=0
                          gamma (int)    : min=48 max=500 step=1 default=57343 value=100
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
                      sharpness (int)    : min=1 max=25 step=1 default=57343 value=12
         backlight_compensation (int)    : min=0 max=2 step=1 default=57343 value=1

```

Henrietta1989, 2019-07-13:
Did you find out what was wrong?

popcornmix, 2019-07-16:
Not here. Try searching in the linked linux issue tracker. This isn't the right place for kernel bug reports.

closed 2019-07-16 by popcornmix