Hexxeh/rpi-firmware #181: Kernel: making uas module built-in

vpelletier, 2018-07-13:
Being tired of burning through micro µSD cards and wishing to keep my pi stand-alone, I purchased a UAS-enabled USB enclosure and a (cheap) SSD, and chain-boot on it (booloader & kernel are still on sdcard fat partition, with `root=PARTUUID=<an ssd partition>`). It boots fine, but kernel uses USB mass-storage module.

Mass storage does not give access to some features, like trim. UAS (USB-attached SCSI) protocol, on the other hand, does give access to trim, so I would like to be able to boot from it. Would it be possible to build kernel images with `CONFIG_USB_UAS=Y` ? (building as a module would be no good without and initrd)

pelwell, 2018-07-13:
`CONFIG_USB_UAS=m` creates a not insignificant (but not heavyweight) 29KB module. I'm not keen to add ~29KB to everybody's kernel so you can trim your SSD using a stock RPi kernel. If you don't think the impact would be so great, build the kernel with and without this option and report the output of `free` in both.

satmandu, 2020-01-15:
Changing one's setup to use an initramfs doesn't seem to be too hard: https://raspberrypi.stackexchange.com/a/103849