Hexxeh/rpi-firmware #182: OpenMax hardware encoder broken since 4.14.58. OK with 4.14.56

Serveurperso, 2018-08-12:
Hello, Sorry for my english (hello from Paris) I use FFMpeg to hardware encode H.264 video. I found the exact buggy commit. I use the last FFMpeg from source master.

It's Okay with the  4.14.56 : https://github.com/Hexxeh/rpi-firmware/commit/d985893ae67195d0cce632efe4437e5fcde4b64b
It's Broken with the 4.14.58 : https://github.com/Hexxeh/rpi-firmware/commit/f1eab5a34dda932ca1ad447bdb35edb0cb55263c

(I use rpi-update [commit hash] + reboot to swap my kernel, awesome cool tool)

Default loglevel :
```
#Please note the cmdline is a ffmpeg symlink

(root|/var/www) /var/www/processdiffusion -f v4l2 -i /dev/video0 -pix_fmt yuv420p -codec:v h264_omx -profile:v baseline -b:v 1400000 -maxrate 1400000 -bufsize 1400000 -f rawvideo tcp://127.0.0.1:8004
ffmpeg version git-2018-08-11-00ae349 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armhf --enable-gpl --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi
  libavutil      56. 19.100 / 56. 19.100
  libavcodec     58. 22.101 / 58. 22.101
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 26.100 /  7. 26.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 211.282037, bitrate: 165888 kb/s
    Stream #0:0: Video: rawvideo (YUY2 / 0x32595559), yuyv422, 720x576, 165888 kb/s, 25 fps, 25 tbr, 1000k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
[h264_omx @ 0x2c09210] Using OMX.broadcom.video_encode
[h264_omx @ 0x2c09210] err 80001005 (-2147479547) on line 481
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
Conversion failed!
```

Debug loglevel :
```
#Please note the cmdline is a ffmpeg symlink

(root|~) /var/www/processdiffusion -loglevel debug -f v4l2 -i /dev/video0 -pix_fmt yuv420p -codec:v h264_omx -profile:v baseline -b:v 1400000 -maxrate 1400000 -bufsize 1400000 -f rawvideo tcp://127.0.0.1:8004
ffmpeg version git-2018-08-11-00ae349 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armhf --enable-gpl --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi
  libavutil      56. 19.100 / 56. 19.100
  libavcodec     58. 22.101 / 58. 22.101
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 26.100 /  7. 26.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'v4l2'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-codec:v' ... matched as option 'codec' (codec name) with argument 'h264_omx'.
Reading option '-profile:v' ... matched as option 'profile' (set profile) with argument 'baseline'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '1400000'.
Reading option '-maxrate' ... matched as AVOption 'maxrate' with argument '1400000'.
Reading option '-bufsize' ... matched as AVOption 'bufsize' with argument '1400000'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option 'tcp://127.0.0.1:8004' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Applying option f (force format) with argument v4l2.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x1df6300] fd:4 capabilities:84200001
[video4linux2,v4l2 @ 0x1df6300] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x1df6300] Querying the device for the current frame size
[video4linux2,v4l2 @ 0x1df6300] Setting frame size to 720x576
[video4linux2,v4l2 @ 0x1df6300] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1df6300] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1df6300] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1df6300] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1df6300] The V4L2 driver changed the pixel format from 0x32315659 to 0x56595559
[video4linux2,v4l2 @ 0x1df6300] Trying to set codec:rawvideo pix_fmt:yuv422p
[video4linux2,v4l2 @ 0x1df6300] The V4L2 driver changed the pixel format from 0x50323234 to 0x56595559
[video4linux2,v4l2 @ 0x1df6300] Trying to set codec:rawvideo pix_fmt:yuyv422
[video4linux2,v4l2 @ 0x1df6300] All info found
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 602.135120, bitrate: 165888 kb/s
    Stream #0:0, 1, 1/1000000: Video: rawvideo, 1 reference frame (YUY2 / 0x32595559), yuyv422, 720x576, 0/1, 165888 kb/s, 25 fps, 25 tbr, 1000k tbn, 1000k tbc
Successfully opened the file.
Parsing a group of options: output url tcp://127.0.0.1:8004.
Applying option pix_fmt (set pixel format) with argument yuv420p.
Applying option codec:v (codec name) with argument h264_omx.
Applying option profile:v (set profile) with argument baseline.
Applying option b:v (video bitrate (please use -b:v)) with argument 1400000.
Applying option f (force format) with argument rawvideo.
Successfully parsed a group of options.
Opening an output file: tcp://127.0.0.1:8004.
[tcp @ 0x1df9dc0] No default whitelist set
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[rawvideo @ 0x1df7ac0] PACKET SIZE: 829440, STRIDE: 1440
detected 4 logical cores
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'video_size' to value '720x576'
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'pix_fmt' to value '1'
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'time_base' to value '1/1000000'
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'pixel_aspect' to value '0/1'
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'sws_param' to value 'flags=2'
[graph 0 input from stream 0:0 @ 0x1e03660] Setting 'frame_rate' to value '25/1'
[graph 0 input from stream 0:0 @ 0x1e03660] w:720 h:576 pixfmt:yuyv422 tb:1/1000000 fr:25/1 sar:0/1 sws_param:flags=2
[format @ 0x1e03ad0] Setting 'pix_fmts' to value 'yuv420p'
[auto_scaler_0 @ 0x1e03840] Setting 'flags' to value 'bicubic'
[auto_scaler_0 @ 0x1e03840] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x1e03ad0] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
[AVFilterGraph @ 0x1e02e40] query_formats: 4 queried, 2 merged, 1 already done, 0 delayed
[auto_scaler_0 @ 0x1e03840] w:720 h:576 fmt:yuyv422 sar:0/1 -> w:720 h:576 fmt:yuv420p sar:0/1 flags:0x4
[h264_omx @ 0x1df9220] Using OMX.broadcom.video_encode
[h264_omx @ 0x1df9220] err 80001005 (-2147479547) on line 481
Error initializing output stream 0:0 -- Error while opening encoder for output stream #0:0 - maybe incorrect parameters such as bit_rate, rate, width or height
[AVIOContext @ 0x1e01e50] Statistics: 0 seeks, 0 writeouts
Conversion failed!
(root|~)
```

Thanks,
Pascal

popcornmix, 2018-08-12:
Have you tried a more recent firmware?
https://github.com/Hexxeh/rpi-firmware/commit/be79970ac6a9975b5b617bc2d3e45b83d882c8c9
includes an alignment fix which may help.

Serveurperso, 2018-08-12:
Yes of course I tested few other and the last : all are NotOK

popcornmix, 2018-08-12:
Ping @6by9

6by9, 2018-08-13:
Please provide the logging from the working firmware.

Direct support for YUYV was added recently (ec9d84e) so previously ffmpeg would have been converting from YUYV to YU12 on the CPU before passing to video_encode, and indeed it appears to be doing so via auto_scalar.
You may as well try removing the ```-pix_fmt yuv420p``` to try and make use of the native support.

Line 481 is the SetParameter(OMX_IndexParamPortDefinition) call, so there appears to be something that it has decided it doesn't like. 
The requirements on YUV420PackedPlanar shouldn't have changed - I'm expecting the stride to be a multiple of 32 bytes, which 720 isn't (736 is). The change was written with MMAL in mind. I wonder if the old IL code allowed a multiple of 16. I'll check later.

I have a test rig that can duplicate this fairly easily, so I'll give it a try when I get a moment.

Serveurperso, 2018-08-13:
Here is the working ffmpeg debug log (an online live robot with 4.14.56)

```
(root|~) /var/www/processdiffusion -loglevel debug -f v4l2 -i /dev/video0 -pix_fmt yuv420p -codec:v h264_omx -profile:v baseline -b:v 1400000 -maxrate 1400000 -bufsize 1400000 -f rawvideo tcp://127.0.0.1:8004 > /var/www/temp/github.txt
ffmpeg version git-2018-08-11-00ae349 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armhf --enable-gpl --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi
  libavutil      56. 19.100 / 56. 19.100
  libavcodec     58. 22.101 / 58. 22.101
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 26.100 /  7. 26.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'v4l2'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-pix_fmt' ... matched as option 'pix_fmt' (set pixel format) with argument 'yuv420p'.
Reading option '-codec:v' ... matched as option 'codec' (codec name) with argument 'h264_omx'.
Reading option '-profile:v' ... matched as option 'profile' (set profile) with argument 'baseline'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '1400000'.
Reading option '-maxrate' ... matched as AVOption 'maxrate' with argument '1400000'.
Reading option '-bufsize' ... matched as AVOption 'bufsize' with argument '1400000'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option 'tcp://127.0.0.1:8004' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Applying option f (force format) with argument v4l2.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x19e5300] fd:4 capabilities:84200001
[video4linux2,v4l2 @ 0x19e5300] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x19e5300] Querying the device for the current frame size
[video4linux2,v4l2 @ 0x19e5300] Setting frame size to 720x576
[video4linux2,v4l2 @ 0x19e5300] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x19e5300] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x19e5300] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x19e5300] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x19e5300] The V4L2 driver changed the pixel format from 0x32315659 to 0x56595559
[video4linux2,v4l2 @ 0x19e5300] Trying to set codec:rawvideo pix_fmt:yuv422p
[video4linux2,v4l2 @ 0x19e5300] The V4L2 driver changed the pixel format from 0x50323234 to 0x56595559
[video4linux2,v4l2 @ 0x19e5300] Trying to set codec:rawvideo pix_fmt:yuyv422
[video4linux2,v4l2 @ 0x19e5300] All info found
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 30992.232076, bitrate: 165888 kb/s
    Stream #0:0, 1, 1/1000000: Video: rawvideo, 1 reference frame (YUY2 / 0x32595559), yuyv422, 720x576, 0/1, 165888 kb/s, 25 fps, 25 tbr, 1000k tbn, 1000k tbc
Successfully opened the file.
Parsing a group of options: output url tcp://127.0.0.1:8004.
Applying option pix_fmt (set pixel format) with argument yuv420p.
Applying option codec:v (codec name) with argument h264_omx.
Applying option profile:v (set profile) with argument baseline.
Applying option b:v (video bitrate (please use -b:v)) with argument 1400000.
Applying option f (force format) with argument rawvideo.
Successfully parsed a group of options.
Opening an output file: tcp://127.0.0.1:8004.
[tcp @ 0x19e8dc0] No default whitelist set
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[rawvideo @ 0x19e6ac0] PACKET SIZE: 829440, STRIDE: 1440
detected 4 logical cores
[graph 0 input from stream 0:0 @ 0x19f2660] Setting 'video_size' to value '720x576'
[graph 0 input from stream 0:0 @ 0x19f2660] Setting 'pix_fmt' to value '1'
[... 76 lines cut ...]
[rawvideo @ 0x19e6ac0] PACKET SIZE: 829440, STRIDE: 1440.72 bitrate= 120.0kbits/s dup=5 drop=4 speed=1.06x
    Last message repeated 10 times
*** dropping frame 104 from stream 0 at ts 102
[rawvideo @ 0x19e6ac0] PACKET SIZE: 829440, STRIDE: 1440
    Last message repeated 2 times
[rawvideo @ 0x19e6ac0] PACKET SIZE: 829440, STRIDE: 1440.24 bitrate= 117.7kbits/s dup=5 drop=5 speed=1.05x
    Last message repeated 5 times
[h264_omx @ 0x19e8220] OMX event 4 c9 1
frame=  114 fps= 27 q=-0.0 Lsize=      67kB time=00:00:04.56 bitrate= 119.8kbits/s dup=5 drop=5 speed=1.06x
video:67kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (/dev/video0):
  Input stream #0:0 (video): 114 packets read (94556160 bytes); 114 frames decoded; 
  Total: 114 packets (94556160 bytes) demuxed
Output file #0 (tcp://127.0.0.1:8004):
  Output stream #0:0 (video): 114 frames encoded; 115 packets muxed (68297 bytes); 
  Total: 115 packets (68297 bytes) muxed
114 frames successfully decoded, 0 decoding errors
[AVIOContext @ 0x19f0e50] Statistics: 0 seeks, 115 writeouts
[h264_omx @ 0x19e8220] OMX state changed to 2
[h264_omx @ 0x19e8220] OMX state changed to 1
Exiting normally, received signal 2.
(root|~)

```
(Raspberry PI 3+ on "Ptibot" robot live here https://www.serveurperso.com/?page=robot)

Serveurperso, 2018-08-13:
I am at the limit of the USB2 bandwidth because I have glitches with the sound only when the video is running at this optimal resolution. all optimization idea are welcome.

6by9, 2018-08-13:
Thanks. I'll look at those logs later.
I have just recalled that the hardware block I've switched to using **requires** the stride to be aligned to a multiple of 32, but I guess the old software conversion didn't.
It'll be a bit nasty, but I'll look into how difficult it will be to switch back to the old software routines if the format doesn't fulfil the requirements for the hardware route.

uvcvideo (the kernel driver for USB Video Class devices) is not the prettiest, and has to do a memcpy of the incoming frames from the USB subsystem URB buffers into the V4L2 buffers (https://elixir.bootlin.com/linux/latest/source/drivers/media/usb/uvc/uvc_video.c#L1070).
This tripped up a previous project I worked on, and at the extreme results in the USB side waiting for URBs to be returned and therefore dropping frames. IIRC There is a module parameter to allocate additional URBs for the device, and that solved the problem for us.
...
Ah no, it's a define (I suspect we made it a module parameter). https://elixir.bootlin.com/linux/latest/source/drivers/media/usb/uvc/uvcvideo.h#L169

Serveurperso, 2018-08-13:
You say I need to recompile the kernel with more URBs ? Can you suggest a tuning to improve my uvcvideo performance ? Or what I can use as alternative for USB capture ? I need a bit of performance to get a perfect system.

```
/* Number of isochronous URBs. */
#define UVC_URBS		5
/* Maximum number of packets per URB. */
#define UVC_MAX_PACKETS		32
```

6by9, 2018-08-13:
I can't give a definitive answer as I'm not the expert on the USB side. I was more relaying an experience on a totally different platform where a UVC camera (actually USB3) was throwing data into the stack faster than it could be consumed as it was highly bursty, and we needed to increase UVC_URBs (and recompile the kernel). You'd need to benchmark that memcpy to confirm whether that is the issue in your case.

There is also a limitation of the Pi USB controller that it doesn't handle isochronous transfers brilliantly.

The Pi supports hardware accelerated MJPEG via IL or MMAL, but FFMPEG doesn't.
Having said that, the MMAL decoder would probably only need a half-dozen lines of code added to support it, mainly setting up ```format_in->encoding``` in ```ffmmal_init_decoder```, and a ```FFMMAL_DEC``` definition at the end of the file. (libavcodec/mmaldec.c).
I'm not sure how well FFMPEG differentiates between MJPEG and JPEG, so there may be minor issues. A couple of ffmpeg-dev list posts recently have been about adding some MJPEG codecs and were discussing it.

Having just tried rebuilding FFMPEG I find that the IL encode support code only supports YU12/I420/yuv420p, so the fact that the firmware now supports YUYV is irrelevant and will always result in a format conversion on the ARM.

Not off the shelf, but there is an option to use the Analog Devices ADV7280-MA or ADV7282-MA chips to stream the data via the CSI-2 camera interface. They also include a hardware deinterlacer. Several forum threads about using it.

Serveurperso, 2018-08-13:
I tested two PI cam "emulators" or Analog to CSI converters breakout/boards without success (too unreliable drivers) or NTSC only input (I need PAL) ... :(

yuv420p conversion is not a problem, I always worked with yuv420p in my FFMPEG cmdline there is no more latency.

if I can get a hardware MJPEG input I think it's the best way.

Serveurperso, 2018-08-13:
https://github.com/FFmpeg/FFmpeg/blob/master/libavcodec/mmaldec.c
Yes there is no MJPEG init inside mmaldec.c :( I must try to work on it.
```
    format_in = decoder->input[0]->format;
    format_in->type = MMAL_ES_TYPE_VIDEO;
    switch (avctx->codec_id) {
    case AV_CODEC_ID_MPEG2VIDEO:
        format_in->encoding = MMAL_ENCODING_MP2V;
        break;
    case AV_CODEC_ID_MPEG4:
        format_in->encoding = MMAL_ENCODING_MP4V;
        break;
    case AV_CODEC_ID_VC1:
        format_in->encoding = MMAL_ENCODING_WVC1;
        break;
    case AV_CODEC_ID_H264:
    default:
        format_in->encoding = MMAL_ENCODING_H264;
        break;
    }
```

6by9, 2018-08-13:
If you're referring to https://lintestsystems.com/products/picapture-sd1 then that is NOT the device I'm talking about. 

We now have a Pi kernel driver for the CSI-2 receiver, and there is a Linux kernel driver for the ADV7180 family of chips that includes the 7280-M and 7282-M. The driver happily copes with PAL and NTSC (you tell V4L2 which one you want), and it spits out UYVY.
This is the device I'm trying to test your use case with, and how come I've now noted that the omx_h264 encoder only supports yuv420p. Mine is a hacked up version of the [http://www.analog.com/media/en/technical-documentation/user-guides/EVAL-ADV7282AMEBZ-UG-1175.pdf](EVAL-ADV7282AMEBZ board), with SMA tails to the Pi camera connector. As I said, it is not available as an off-the-shelf option at the moment.

UYVY to yuv420p is not that heavyweight an operation (at least compared to conversion to RGB), but it is still a step that is accessing the every single image pixel. Memory bandwidth needs to be considered as much as CPU processing power.

6by9, 2018-08-13:
If you're referencing the Auvidea B101 (or B102) as HDMI to CSI, then that too is now using the Linux kernel driver rather than the closed source firmware. It will also capture the HDMI audio over I2S if you wire that bit up too.
Because of the way V4L2 works there are a couple of hoops to jump through in providing an EDID to the device, and then selecting the correct dv timings mode.

Search the forums for TC358743 (the chip used on those boards).

Serveurperso, 2018-08-13:
Yes I have a TC358743 Auvidea B101/2 with I2S audio. I must re-check it. I successfully used last year with the bad driver.
I found new cheap FEBON USB2 adapter with hardware deinterlacer but there is always this bandwidth limitation.

Serveurperso, 2018-08-13:
I already tested H.264 to H.264 transcoding with MMAL because my capture card can do H.264 high profile directly. I need Baseline for Broadway (web browser) but the MMAL decoder part give to mutch latency. I think MJPEG to RAW to H264 can be way better (I need < 0.5s latency)...

Serveurperso, 2018-08-13:
Else a cheap Intel board + cheap chineese capture card + full software stack from FFmpeg do a perfect job for cheaper (I get <0.5s latency audio+video from camera to browser!)...

Serveurperso, 2018-08-14:
I get frames duplicate&drop even with a low resolution like 320x240 (25FPS) ! There is a huge performance problem with USB on raspberry PI 3+ (or a FFmpeg bug!)

This is another issue need a lot more testing (with older versions of kernel and/or FFmpeg) but I post the log here.

```

(root|~) /var/www/processdiffusion -loglevel debug -f v4l2 -framerate 25 -video_size 320x240 -i /dev/video0 -codec:v h264_omx -profile:v baseline -b:v 1400000 -maxrate 1400000 -bufsize 1400000 -f rawvideo tcp://127.0.0.1:8004
ffmpeg version git-2018-08-11-00ae349 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
  configuration: --arch=armhf --enable-gpl --enable-nonfree --enable-mmal --enable-omx --enable-omx-rpi
  libavutil      56. 19.100 / 56. 19.100
  libavcodec     58. 22.101 / 58. 22.101
  libavformat    58. 17.101 / 58. 17.101
  libavdevice    58.  4.101 / 58.  4.101
  libavfilter     7. 26.100 /  7. 26.100
  libswscale      5.  2.100 /  5.  2.100
  libswresample   3.  2.100 /  3.  2.100
  libpostproc    55.  2.100 / 55.  2.100
Splitting the commandline.
Reading option '-loglevel' ... matched as option 'loglevel' (set logging level) with argument 'debug'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'v4l2'.
Reading option '-framerate' ... matched as AVOption 'framerate' with argument '25'.
Reading option '-video_size' ... matched as AVOption 'video_size' with argument '320x240'.
Reading option '-i' ... matched as input url with argument '/dev/video0'.
Reading option '-codec:v' ... matched as option 'codec' (codec name) with argument 'h264_omx'.
Reading option '-profile:v' ... matched as option 'profile' (set profile) with argument 'baseline'.
Reading option '-b:v' ... matched as option 'b' (video bitrate (please use -b:v)) with argument '1400000'.
Reading option '-maxrate' ... matched as AVOption 'maxrate' with argument '1400000'.
Reading option '-bufsize' ... matched as AVOption 'bufsize' with argument '1400000'.
Reading option '-f' ... matched as option 'f' (force format) with argument 'rawvideo'.
Reading option 'tcp://127.0.0.1:8004' ... matched as output url.
Finished splitting the commandline.
Parsing a group of options: global .
Applying option loglevel (set logging level) with argument debug.
Successfully parsed a group of options.
Parsing a group of options: input url /dev/video0.
Applying option f (force format) with argument v4l2.
Successfully parsed a group of options.
Opening an input file: /dev/video0.
[video4linux2,v4l2 @ 0x1b19370] fd:4 capabilities:84200001
[video4linux2,v4l2 @ 0x1b19370] Current input_channel: 0, input_name: Camera 1, input_std: 0
[video4linux2,v4l2 @ 0x1b19370] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1b19370] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1b19370] The V4L2 driver changed the pixel format from 0x32315559 to 0x56595559
[video4linux2,v4l2 @ 0x1b19370] Trying to set codec:rawvideo pix_fmt:yuv420p
[video4linux2,v4l2 @ 0x1b19370] The V4L2 driver changed the pixel format from 0x32315659 to 0x56595559
[video4linux2,v4l2 @ 0x1b19370] Trying to set codec:rawvideo pix_fmt:yuv422p
[video4linux2,v4l2 @ 0x1b19370] The V4L2 driver changed the pixel format from 0x50323234 to 0x56595559
[video4linux2,v4l2 @ 0x1b19370] Trying to set codec:rawvideo pix_fmt:yuyv422
[video4linux2,v4l2 @ 0x1b19370] Setting time per frame to 1/25
[video4linux2,v4l2 @ 0x1b19370] All info found
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 259.225212, bitrate: 30720 kb/s
    Stream #0:0, 1, 1/1000000: Video: rawvideo, 1 reference frame (YUY2 / 0x32595559), yuyv422, 320x240, 0/1, 30720 kb/s, 25 fps, 25 tbr, 1000k tbn, 1000k tbc
Successfully opened the file.
Parsing a group of options: output url tcp://127.0.0.1:8004.
Applying option codec:v (codec name) with argument h264_omx.
Applying option profile:v (set profile) with argument baseline.
Applying option b:v (video bitrate (please use -b:v)) with argument 1400000.
Applying option f (force format) with argument rawvideo.
Successfully parsed a group of options.
Opening an output file: tcp://127.0.0.1:8004.
[tcp @ 0x1b1ce30] No default whitelist set
Successfully opened the file.
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_omx))
Press [q] to stop, [?] for help
cur_dts is invalid (this is harmless if it occurs once at the start per stream)
[rawvideo @ 0x1b1ab60] PACKET SIZE: 153600, STRIDE: 640
detected 4 logical cores
[... 85 lines cut ...]
    Last message repeated 12 times
[rawvideo @ 0x1b1ab60] PACKET SIZE: 153600, STRIDE: 6404.28 bitrate=  17.0kbits/s dup=5 drop=4 speed=1.06x
    Last message repeated 1 times
*** dropping frame 109 from stream 0 at ts 107
[rawvideo @ 0x1b1ab60] PACKET SIZE: 153600, STRIDE: 640
    Last message repeated 10 times
[rawvideo @ 0x1b1ab60] PACKET SIZE: 153600, STRIDE: 6404.76 bitrate=  17.0kbits/s dup=5 drop=5 speed=1.05x
    Last message repeated 3 times
[h264_omx @ 0x1b1c2c0] OMX event 4 c9 1
frame=  124 fps= 26 q=-0.0 Lsize=      11kB time=00:00:04.96 bitrate=  17.8kbits/s dup=5 drop=6 speed=1.05x
video:11kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (/dev/video0):
  Input stream #0:0 (video): 125 packets read (19200000 bytes); 125 frames decoded; 
  Total: 125 packets (19200000 bytes) demuxed
Output file #0 (tcp://127.0.0.1:8004):
  Output stream #0:0 (video): 124 frames encoded; 125 packets muxed (11012 bytes); 
  Total: 125 packets (11012 bytes) muxed
125 frames successfully decoded, 0 decoding errors
[AVIOContext @ 0x1b24ec0] Statistics: 0 seeks, 125 writeouts
[h264_omx @ 0x1b1c2c0] OMX state changed to 2
[h264_omx @ 0x1b1c2c0] OMX state changed to 1
Exiting normally, received signal 2.
(root|~)  

```

Serveurperso, 2018-09-03:
I think it's related... I think...

pelwell, 2018-09-03:
Let me reword that:

Please open a new Issue for a new problem. If you think it may be related to an existing issue then you can say so in the new Issue.

6by9, 2018-09-03:
Raspivid doesn't use OpenMax (it uses MMAL), therefore it isn't directly related. New issue please.

Serveurperso, 2018-09-03:
I understood the first message despite the approximate answer
I already write the new issue.
Bad english, Hello from Paris

Serveurperso, 2018-09-03:
https://github.com/Hexxeh/rpi-firmware/issues/185

Ruffio, 2020-08-26:
@Serveurperso can this be closed?

Serveurperso, 2020-08-26:
Hello, no problem since a long time:) This is closed. We run 24/7 many robots with camera on the last Buster image fully updated to last stable kernel (no rpi-update) and Raspberry PI 2/3A/3B/4 with all supported sensors v1 v2 and PRO official camera.. (https://www.vigibot.com)
I can make specific rpi-update version testing if you want.

Ruffio, 2020-08-26:
@Serveurperso can you close it then? I can't...

closed 2020-08-26 by Serveurperso