Hexxeh/rpi-firmware #184: kernel 4.14.66 cannot reboot

erchegyia, 2018-08-26:
After updating (66bb2b42f07e2495b54804385eaed593ee851cd1) reboot always fails with "Fatal exception in interrupt". [Screen](https://photos.app.goo.gl/YhU5H78m5zqH5jta9) . After cutting the power it boots up without any problem.
I am using stretch.
Rolling back to 4.14.62 (a05e3e55870367362595d5fa531cedbbc75c62dc) solves the issue.

pelwell, 2018-08-28:
Which platform are you using?

Can you grab another screenshot with `framebuffer_height=2160` in config.txt? This should allow us to see more of the kernel dump.

erchegyia, 2018-08-28:
It's Pi 3 Model B + (ARMv8)

I'll come back to you with the extended dump. Is there a way to get the text directly iso making screenshot?

pelwell, 2018-08-28:
I use a serial cable attached to GPIOs 14 & 15 (pins 8 & 10, with ground on pin 6), but that doesn't help you if you don't have one (unless you cross-connect to a second Pi). You would need to add `enable_uart=1` to `config.txt`.

arshad-domautics, 2018-09-25:
i was facing same issue with 4.14.66 and onward version including 4.14.71. I am not sure that it get install all these things in all good condition or not..?
but what i done is, i updated with latest commit 4.14.71 on pi-3 model-B once after "rpi-update" process complete, instead of reboot i have done shutdown ("reboot --poweroff") and when i start manually it worked.

pelwell, 2018-09-25:
I think this issue was another duplicate of https://github.com/raspberrypi/linux/issues/2681 which has been solved, and can therefore be closed.

closed 2018-09-25 by popcornmix