Hexxeh/rpi-firmware #186: Kernel panic

Nokia390, 2018-09-06:
Having kernel panic on cron reboot with 4.14.67, i´ve downgrade to 4.14.62 and all ok.

pelwell, 2018-09-06:
Can you post a log or photo of one of the panics? `framebuffer_height=2160` in config.txt will increase the amount visible on the screen.

Nokia390, 2018-09-06:
I don´t have a monitor conected...i try to find in /var/log, but nothing. where i can find some log, off the error?

pelwell, 2018-09-06:
If the panic happens after the file system is shut down then it won't be recorded anywhere.

How do you know there is a panic without a monitor? The best option might be to use a serial cable attached to 40-way header pins 6, 8 and 10 to capture the panic (if that's what it is) from another device (either another Pi or a PC of some description) - put `enable_uart=1` in config.txt.

sfx2000, 2018-09-06:
Might be a dup of #184 

I've see this on Pi Zero W and Pi 3B+ intermittently - both running headless - rolling back to official works

Odd - nothing exceptional in Linux, wonder if this related to VC4 somehow when passing things over...

Nokia390, 2018-09-07:
Yes, the same (#184). If i cut power it boot normaly. 
After rooling back the firmware it reboot perfect.

sfx2000, 2018-09-12:
Whatever happened - the kernel was pushed to production without a solution, and yes, still seeing panics...

@pelwell @popcornmix - might want to dig deeper into this - headless folks got the production push, and you might end up with a lot of Pi's stuck on reboot.

sfx2000, 2018-09-12:
> Yes, the same (#184). If i cut power it boot normaly.
> After rooling back the firmware it reboot perfect.

That's pretty much my observation as well - most of the changes shouldn't impact things, but it's hard to debug as there's no debug info to share...

mrplumber, 2018-09-14:
Hi.
Confirming the problem on my two B3 RPIs. Both running Raspbian Light headless.
```
uname -srvm
Linux 4.14.69-v7+ #1141 SMP Mon Sep 10 15:26:29 BST 2018 armv7l

```
Kernel panics or oopsies happen randomly on reboot or poweroff.  As suggested above added `framebuffer_height=2160` in config.txt and made some screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

sfx2000, 2018-09-14:
May be the same issue as this...

https://github.com/raspberrypi/linux/issues/2680

kapitainsky, 2018-09-15:
I have noticed the same problem recently - running headless so was not sure what is wrong. Happens both with RPi 2 and 3.

Bootpanic, 2018-09-16:
> May be the same issue as this...
> 
> [raspberrypi/linux#2680](https://github.com/raspberrypi/linux/issues/2680)

Seems to be the same as https://github.com/raspberrypi/linux/issues/2681

closed 2018-09-17 by popcornmix