Hexxeh/rpi-firmware #187: cifs filesystem not supported by the system

dn4hc, 2018-09-23:
I'm not able to mount SMB after the last kernel update

`pi@pihole:~ $ sudo mount -t cifs //R7000.local/sdb1 /mnt/R7000 --verbose -o username=guest,password=guest,domain=workgroup,iocharset=u      tf8,sec=ntlm,vers=1.0
mount.cifs kernel mount options: ip=10.1.1.1,unc=\\R7000.local\sdb1,iocharset=utf8,sec=ntlm,vers=1.0,user=guest,domain=workgroup,pass=      ********
mount error: cifs filesystem not supported by the system
mount error(19): No such device
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)
pi@pihole:~ $ cat /proc/filesystems
nodev   sysfs
nodev   rootfs
nodev   ramfs
nodev   bdev
nodev   proc
nodev   cpuset
nodev   cgroup
nodev   cgroup2
nodev   tmpfs
nodev   devtmpfs
nodev   configfs
nodev   debugfs
nodev   tracefs
nodev   sockfs
nodev   pipefs
nodev   rpc_pipefs
nodev   devpts
        ext3
        ext2
        ext4
        vfat
        msdos
nodev   nfs
nodev   nfs4
nodev   autofs
        f2fs
nodev   mqueue
pi@pihole:~ $ sudo modprobe configs
pi@pihole:~ $ sudo modprobe cifs
modprobe: ERROR: could not insert 'cifs': Exec format error
pi@pihole:~ $ uname -a
Linux pihole 4.14.70-v7+ #1144 SMP Tue Sep 18 17:34:46 BST 2018 armv7l GNU/Linux
`

pelwell, 2018-09-23:
modprobe cifs works for me, so I think your filesystem or the download is corrupt. What do the following commands return (watch the backtick quote characters):
```
pi@raspberrypi:$ uname -a
pi@raspberrypi:$ md5sum /lib/modules/`uname -r`/kernel/fs/cifs/cifs.ko
```

pelwell, 2018-09-23:
The correct md5sum for the current `cifs` module begins `6cf989fe`... - if you get a different result then retry the rpi-update after first deleting `/boot/.firmware_revision`.

dn4hc, 2018-09-23:
I'm getting different result. 

pi@pihole:~ $ sudo md5sum /lib/modules/`uname -r`/ke
rnel/fs/cifs/cifs.ko
29bec25eeb611672a59fdb522cd5469e  /lib/modules/4.14.
70-v7+/kernel/fs/cifs/cifs.ko

pelwell, 2018-09-23:
The current firmware includes the 4.14.71 kernel, so you are behind (or you haven't rebooted after the update). Reboot, run the commands again, then rpi-update.

dn4hc, 2018-09-23:
Retry of rpi-update worked. Thanks