Hexxeh/rpi-firmware #191: error when updating sdk

crazycatdevs, 2018-11-29:
Hello,
I'm trying to  update my RPi and I have an error, I'm not the only one (cf https://forum.raspfr.org/Probleme-rpi-update-t-2355.html):
`root@raspbian:~# rpi-update

 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Backing up modules 4.14.70-v7+
#############################################################
This update bumps to rpi-4.14.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=197689
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    322      0 --:--:-- --:--:-- --:--:--   322
100 53.0M  100 53.0M    0     0   832k      0  0:01:05  0:01:05 --:--:--  852k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.14.83+
 *** depmod 4.14.83-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
cp: impossible d'évaluer '//root/.rpi-firmware/vc/sdk/*': Aucun fichier ou dossier de ce type`

Looking in **/root/.rpi-firmware/vc/**, there's only **hardfp/opt/vc/** inside, no sdk/ directory.

kapitainsky, 2018-11-29:
the same issue updating from the previous 4.14.82-v7+ to 4.14.83-v7+

popcornmix, 2018-11-29:
Sorry. Something is wrong. I've reverted the update (which should fix error) and will push again when fixed.

popcornmix, 2018-11-29:
Should be fixed now and updated kernel pushed. Please test and confirm.

kapitainsky, 2018-11-29:
works now. no issues

crazycatdevs, 2018-11-29:
Confirmed: works perfectly.

closed 2018-11-29 by crazycatdevs