Hexxeh/rpi-firmware #195: raspberrypi 3b kernel Internal error: Oops: 17 [#1] SMP ARM for BCM2835

cxueqin, 2019-01-05:
Hi friend:
   I use raspberrypi 3B board as a  wifi station mode to connect to an AP. The wifi connection was made by wpa_supplicant with 802.1X authtication params.  The 802.1x params list below:
   * key_mgmt : WPA-EAP
   * eap: PEAP
   * proto: WPA RSN
   * pairwise: CCMP
   * group: CCMP
   * NetworkIdentity
   * password

 The wifi connection was fine at first. I used wpa_cli to reassociate the AP with the same params every two minutes. In other words, the rp 3b board test 802.1x access authentication every two minutes. After hundreds of successful reassociations, the kernel went wrong. The syslog shown

> Jan  4 07:14:40 NetFrog kernel: [21393.314945] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
> Jan  4 07:14:40 NetFrog kernel: [21393.315365] brcmfmac: brcmf_cfg80211_escan: Connecting: status (3)
> Jan  4 07:14:40 NetFrog kernel: [21393.315375] brcmfmac: brcmf_cfg80211_scan: scan error (-11)
> Jan  4 07:14:45 NetFrog kernel: [21398.610986] ------------[ cut here ]------------
> Jan  4 07:14:45 NetFrog kernel: [21398.611241] WARNING: CPU: 0 PID: 42 at net/wireless/sme.c:752 __cfg80211_connect_result+0x398/0x3e4 [cfg80211]
> Jan  4 07:14:45 NetFrog kernel: [21398.611248] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd asix uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
> Jan  4 07:14:45 NetFrog kernel: [21398.611376] CPU: 0 PID: 42 Comm: kworker/u8:1 Tainted: G        WC      4.14.90-v7+ #1183
> Jan  4 07:14:45 NetFrog kernel: [21398.611381] Hardware name: BCM2835
> Jan  4 07:14:45 NetFrog kernel: [21398.611571] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
> Jan  4 07:14:45 NetFrog kernel: [21398.611605] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (show_stack+0x20/0x24)
> Jan  4 07:14:45 NetFrog kernel: [21398.611622] [<8010c174>] (show_stack) from [<8078b864>] (dump_stack+0xd4/0x118)
> Jan  4 07:14:45 NetFrog kernel: [21398.611639] [<8078b864>] (dump_stack) from [<8011dd70>] (__warn+0xf8/0x110)
> Jan  4 07:14:45 NetFrog kernel: [21398.611654] [<8011dd70>] (__warn) from [<8011de58>] (warn_slowpath_null+0x30/0x38)
> Jan  4 07:14:45 NetFrog kernel: [21398.611850] [<8011de58>] (warn_slowpath_null) from [<7f22ed4c>] (__cfg80211_connect_result+0x398/0x3e4 [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612227] [<7f22ed4c>] (__cfg80211_connect_result [cfg80211]) from [<7f202ec4>] (cfg80211_process_wdev_events+0x10c/0x14c [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612598] [<7f202ec4>] (cfg80211_process_wdev_events [cfg80211]) from [<7f202f40>] (cfg80211_process_rdev_events+0x3c/0x70 [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.612965] [<7f202f40>] (cfg80211_process_rdev_events [cfg80211]) from [<7f1fd294>] (cfg80211_event_work+0x24/0x2c [cfg80211])
> Jan  4 07:14:45 NetFrog kernel: [21398.613158] [<7f1fd294>] (cfg80211_event_work [cfg80211]) from [<801379f0>] (process_one_work+0x158/0x454)
> Jan  4 07:14:45 NetFrog kernel: [21398.613173] [<801379f0>] (process_one_work) from [<80137d50>] (worker_thread+0x64/0x5b8)
> Jan  4 07:14:45 NetFrog kernel: [21398.613187] [<80137d50>] (worker_thread) from [<8013ddd4>] (kthread+0x13c/0x16c)
> Jan  4 07:14:45 NetFrog kernel: [21398.613205] [<8013ddd4>] (kthread) from [<801080ac>] (ret_from_fork+0x14/0x28)
> Jan  4 07:14:45 NetFrog kernel: [21398.613293] ---[ end trace 5eac627de54b823b ]---
> Jan  4 07:14:46 NetFrog kernel: [21399.528228] Unable to handle kernel NULL pointer dereference at virtual address 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528253] pgd = a50b0000
> Jan  4 07:14:46 NetFrog kernel: [21399.528264] [00000000] *pgd=373d8835, *pte=00000000, *ppte=00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528293] Internal error: Oops: 17 [#1] SMP ARM
> Jan  4 07:14:46 NetFrog kernel: [21399.528304] Modules linked in: fuse rfcomm cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic brcmfmac brcmutil cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd asix uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
> Jan  4 07:14:46 NetFrog kernel: [21399.528432] CPU: 1 PID: 1176 Comm: wpa_supplicant Tainted: G        WC      4.14.90-v7+ #1183
> Jan  4 07:14:46 NetFrog kernel: [21399.528442] Hardware name: BCM2835
> Jan  4 07:14:46 NetFrog kernel: [21399.528453] task: b674cb00 task.stack: a790e000
> Jan  4 07:14:46 NetFrog kernel: [21399.528476] PC is at call_commit_handler.part.2+0x18/0x44
> Jan  4 07:14:46 NetFrog kernel: [21399.528490] LR is at ioctl_standard_call+0x270/0x4bc
> Jan  4 07:14:46 NetFrog kernel: [21399.528502] pc : [<80785b08>]    lr : [<80785da4>]    psr: 20000013
> Jan  4 07:14:46 NetFrog kernel: [21399.528513] sp : a790fde0  ip : a790fdf0  fp : a790fdec
> Jan  4 07:14:46 NetFrog kernel: [21399.528523] r10: b9272800  r9 : 00000034  r8 : 00008b1a
> Jan  4 07:14:46 NetFrog kernel: [21399.528535] r7 : 0000000e  r6 : a790fea0  r5 : 00000003  r4 : 00000002
> Jan  4 07:14:46 NetFrog kernel: [21399.528547] r3 : 00000000  r2 : 60000013  r1 : 60000013  r0 : b9272800
> Jan  4 07:14:46 NetFrog kernel: [21399.528560] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
> Jan  4 07:14:46 NetFrog kernel: [21399.528572] Control: 10c5383d  Table: 250b006a  DAC: 00000055
> Jan  4 07:14:46 NetFrog kernel: [21399.528584] Process wpa_supplicant (pid: 1176, stack limit = 0xa790e210)
> Jan  4 07:14:46 NetFrog kernel: [21399.528595] Stack: (0xa790fde0 to 0xa7910000)
> Jan  4 07:14:46 NetFrog kernel: [21399.528613] fde0: a790fe5c a790fdf0 80785da4 80785afc 806ab2ac a790fe6c a790fe6c a790feb0
> Jan  4 07:14:46 NetFrog kernel: [21399.528630] fe00: 00000000 a536aa80 80697fd0 636ab284 616e6968 6566696c 6669772d 8068bc69
> Jan  4 07:14:46 NetFrog kernel: [21399.528648] fe20: a790fe44 807a4150 a790fe4c a790fe38 807a4150 00008b1a a790fea0 80c7a840
> Jan  4 07:14:46 NetFrog kernel: [21399.528665] fe40: 7efe67b8 00000000 a790e000 00000036 a790fe94 a790fe60 807862e8 80785b40
> Jan  4 07:14:46 NetFrog kernel: [21399.528682] fe60: 7f249c5c 80c7a840 7efe67b8 00008b1a 00000004 00008b1a 80c7a840 7efe67b8
> Jan  4 07:14:46 NetFrog kernel: [21399.528699] fe80: 80c7a840 00000004 a790feec a790fe98 806b59b4 8078613c a790ff14 a790fea8
> Jan  4 07:14:46 NetFrog kernel: [21399.528717] fea0: 6e616c77 00000030 00000000 00000000 7efe67d8 0001000e 00000000 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528734] fec0: a790ff14 00008b1a 0000012a 7efe67b8 80c7a840 00000004 a790e000 00000036
> Jan  4 07:14:46 NetFrog kernel: [21399.528752] fee0: a790ff0c a790fef0 80671330 806b5924 7efe67b8 abf6eaa0 b67fcd80 00008b1a
> Jan  4 07:14:46 NetFrog kernel: [21399.528769] ff00: a790ff7c a790ff10 8029f060 80671218 807a1834 80145efc 00000128 801081c4
> Jan  4 07:14:46 NetFrog kernel: [21399.528787] ff20: a790e000 00000128 a790ff94 a790ff38 808024bc 397d3000 807a1e98 ba361d40
> Jan  4 07:14:46 NetFrog kernel: [21399.528803] ff40: 00000000 00000000 a790fe88 a790e000 a790e010 b67fcd80 7efe67b8 b67fcd80
> Jan  4 07:14:46 NetFrog kernel: [21399.528820] ff60: 00008b1a 00000004 a790e000 00000036 a790ffa4 a790ff80 8029f7bc 8029efc0
> Jan  4 07:14:46 NetFrog kernel: [21399.528839] ff80: 00000016 0000000e 019773c0 7efe67d8 00000036 801081c4 00000000 a790ffa8
> Jan  4 07:14:46 NetFrog kernel: [21399.528855] ffa0: 80108000 8029f784 0000000e 019773c0 00000004 00008b1a 7efe67b8 00000016
> Jan  4 07:14:46 NetFrog kernel: [21399.528872] ffc0: 0000000e 019773c0 7efe67d8 00000036 01989704 00000000 001bc730 019773d0
> Jan  4 07:14:46 NetFrog kernel: [21399.528889] ffe0: 001bcbac 7efe67b4 00141478 76c7680c 20000010 00000004 00000000 00000000
> Jan  4 07:14:46 NetFrog kernel: [21399.528920] [<80785b08>] (call_commit_handler.part.2) from [<80785da4>] (ioctl_standard_call+0x270/0x4bc)
> Jan  4 07:14:46 NetFrog kernel: [21399.528942] [<80785da4>] (ioctl_standard_call) from [<807862e8>] (wext_handle_ioctl+0x1b8/0x20c)
> Jan  4 07:14:46 NetFrog kernel: [21399.528967] [<807862e8>] (wext_handle_ioctl) from [<806b59b4>] (dev_ioctl+0x9c/0x8e8)
> Jan  4 07:14:46 NetFrog kernel: [21399.528992] [<806b59b4>] (dev_ioctl) from [<80671330>] (sock_ioctl+0x124/0x2d8)
> Jan  4 07:14:46 NetFrog kernel: [21399.529016] [<80671330>] (sock_ioctl) from [<8029f060>] (do_vfs_ioctl+0xac/0x7c4)
> Jan  4 07:14:46 NetFrog kernel: [21399.529037] [<8029f060>] (do_vfs_ioctl) from [<8029f7bc>] (SyS_ioctl+0x44/0x6c)
[... cut ...]
  The syslog told kernel Internal error: Oops: 17 [#1] SMP ARM for BCM2835. After that the wpa_supplicant didn't work, the system cpu usage increased gradually which would reach 100 or higher(measured by uptime). And then the linux system didn't work . SSH service couldn't be reach, but icmp ping was still alive.

  I tried the above test several times on the same and different raspberry pi 3B boards. The kernel error problem could be reproduced. The kernel firmware was upgraded to lastest 4.14.90-v7+.

  Any suggestion is welcome. Thank you !

pelwell, 2019-01-06:
https://github.com/raspberrypi/linux/issues/2794 is a better home for this report - please close this issue.

cxueqin, 2019-01-07:
Well,  issue refer to raspberrypi/linux#2794 . This issue is closed. Thanks.

closed 2019-01-07 by cxueqin