Hexxeh/rpi-firmware #199: Any plan for rpi-5.0.y?

artificerpi, 2019-03-09:
As is titled, linux kernel 5.0 has come out, and when will rpi-update support 5.0 kernel?
There's already the `rpi-5.0.y` branch: https://github.com/raspberrypi/linux/tree/rpi-5.0.y

pelwell, 2019-03-09:
5.0 is barely dry - I wouldn't want to use it as my daily driver. The source repo is here for people to build it themselves and kick the tyres, and eventually it _may_ become the source for firmware build. However, new kernel versions are coming thick and fast, and we just jumped from 4.14 to 4.19; it's no coincidence that those two are Long Term Stable versions.

Whobeu, 2019-03-09:
As Linus was quoted as saying, 5.0 would have been 4.21 but he ran out of fingers and toes. I suspect there is not that much in it yet to justify a move.

artificerpi, 2019-03-10:
Got it. But I will build it from source code manually instead to have a try.

closed 2019-03-10 by artificerpi