Hexxeh/rpi-firmware #200: initramfs generation by checking /etc/default/raspberrypi-kernel

dsx724, 2019-03-27:
There's a INITRD=Yes field in /etc/default/raspberrypi-kernel. Is it possible to run the initramfs generation if this is set?

XECDesign, 2019-03-27:
The long term plan is to add an initramfs-tools fork that works with the raspberry pi kernel. You can add your own hooks in the meantime.

dsx724, 2019-03-27:
If a user calls rpi-update, it doesn't call update-initramfs to the best of our knowledge. The machines will break on next boot since it is missing the modules that were included in initramfs.

dsx724, 2019-03-28:
@XECDesign Sorry I filed this ticket on the wrong repo. It was suppose to be in the rpi-update repo. https://github.com/Hexxeh/rpi-update/issues/252

Any chance you can add a call there if you have access to that repo?

XECDesign, 2019-03-28:
> Sorry I filed this ticket on the wrong repo.

No worries, I actually thought I was looking at a different repo when I saw this report as well, so we were both approaching it from different directions.

> Any chance you can add a call there if you have access to that repo?

I do not.

popcornmix, 2019-03-28:
Are you saying it is enough to call ``update-initramfs`` after the kernel has been updated by rpi-update? And this is safe to do for non-initramfs users?

dsx724, 2019-03-28:
@popcornmix I added the follow snippet to line 119 of rpi-update:

`                        if [[ -e ${ROOT_PATH}/etc/default/raspberrypi-kernel ]] && grep -i "^INITRD=yes" ${ROOT_PATH}/etc/default/raspberrypi-kernel > /dev/null; then
                                update-initramfs -c -k "${BASEDIR}"
                                ${ROOT_PATH}/etc/kernel/postinst.d/rpi-initramfs-config "${BASEDIR}"
                        fi
`
rpi-initramfs-config is a script to modify the /boot/config.txt to include the appropriate values for initrd.

`#!/bin/bash
set -e
BOOT_CONFIG_FILE=/boot/config.txt
if [ ! -f "$BOOT_CONFIG_FILE" ]; then
        exit 1
fi

if [ -z "$1" ]; then
        exit 1
fi

pi_list=""

if [ "${1##*-}" = "v7+" ]; then
        pi_list="2 3 3+"
elif [ "${1##*-}" = "v8+" ]; then
        true
else
        pi_list="0 0w 1"
fi
if [ ! -z "$pi_list" ]; then
        config_temp=`mktemp`
        config_temp_bounce=`mktemp`
        cp "$BOOT_CONFIG_FILE" "$config_temp"
        for pi in $pi_list; do
                if grep -E "^\[pi${pi//+/\\+}\]\s*\$" "$config_temp"; then
                        awk -f- "$config_temp" > "$config_temp_bounce" <<EOF
BEGIN { insec = 0; }
/^\\[/ { if (insec) { insec = 0; print "initramfs initrd.img-$1 followkernel"; } }
/^\\[pi${pi//+/\\+}\\]/ { insec = 1; }
/^initramfs initrd\\.img-.* followkernel/ { if (insec) skip = 1; }
/.*/ { if (skip) skip = 0; else print \$0; }
END { if (insec) { insec = 0; print "initramfs initrd.img-$1 followkernel"; } }
EOF
                        cp "$config_temp_bounce" "$config_temp"
                else
                        echo -e "[pi$pi]\ninitramfs initrd.img-$1 followkernel" >> "$config_temp"
                fi
        done
        cat "$config_temp" > "$BOOT_CONFIG_FILE"
        rm "$config_temp" "$config_temp_bounce"
fi
`

dsx724, 2019-03-29:
I added pruning old initrd files as well to prevent the FAT partition from blowing up: https://github.com/dsx724/rpi-update/commit/3e9b44d607793b061ea32c647f69c50dc559e5a3

lurch, 2019-06-08:
@dsx724 I think if you have changes you want to get into rpi-update, then you'll need to submit a PR to https://github.com/Hexxeh/rpi-update/pulls rather than posting commit-links to an issue on https://github.com/Hexxeh/rpi-firmware :wink: 

@popcornmix I think this can be closed as a duplicate of https://github.com/Hexxeh/rpi-update/issues/252

popcornmix, 2019-06-10:
Okay, continue discussion in https://github.com/Hexxeh/rpi-update/issues/252
(or submit a PR).

closed 2019-06-10 by popcornmix