Hexxeh/rpi-firmware #201: Audio stopped working

rien333, 2019-05-01:
After updating my rpi3-B to the newest firmware (`6cf2788e74b497c109865d052cd6fa68d094cf38`) my audio stopped working. For some reason the pi can't find the soundcard anymore:
```bash
$ sudo aplay -l
aplay: device_list:270: no soundcards found...
```
Pulseadio starts fine, but of course won't work without ALSA/a soundcard. I tried changing the common audio related params in `/boot/config.txt` (`dtparam=audio=on/off`, `audio_pwm_mode=0/2`), but to no avail.   Downgrading to `#99c274691c07480762dcda91a0ebfe3c4f519307` fixed the issue for me (picked this version because I figured my audio worked fine around that time)

I'm running the newest Raspbian (9.9), with no updates from `apt-get` available.

pelwell, 2019-05-01:
Duplicate of https://github.com/raspberrypi/linux/issues/2952. Yes, you were first, but their report is in a more appropriate repo and with much more detail.

popcornmix, 2019-05-02:
Can you test again after an rpi-update?

rien333, 2019-05-02:
Works fine now, thanks! (on 10a512752529b2c7970bce43201366521f40e6b3) And sorry for my lack of detail and choosing a somewhat unrelated repo.

closed 2019-05-02 by rien333