Hexxeh/rpi-firmware #203: 4 blinking leds after upgrading Pi 3 to b39dd764176933407df48424f1d0779da05142b1

TauPan, 2019-06-25:
I've seen 768ea94042bade6ea276c6cea358c9645cb5de6a but erroneously used rpi-update as part of my regular upgrade process (some sites recommended this). I'm going to stop this now and that's not really the point here.

I just wanted to point out that both my Pi3s stopped working with a red LED permanently on and a green LED blinking 4 times.

I've managed to rectify this by mounting the sdcard on my laptop and running rpi-update again with 458871abedfd68ea417d24577694dbe9d7d4db4c (the last working version).

If this (breakage on Pi3) is something you expect as the (apparently) ongoing process of adaptation to Pi4, you can just silently ignore this issue.

If you'd like more information from me, I'd gladly provide it.

Kind regards and thanks for your great work
Friedel

popcornmix, 2019-06-25:
When did you update? There was a bug in the rpi-update tool that was fixed at 12:36 today.
If you updated before then, the reason is known (missing start.elf file) and shouldn't occur again.

If it was after then that is something we don't know about.

TauPan, 2019-06-25:
Which time zone? I ran rpi-update before 1030 CEST (UTC +2).

popcornmix, 2019-06-25:
We're in the UK so UTC+1. Sounds like you updated before the bug was fixed.
All should be fine now.

TauPan, 2019-06-25:
Ok, thanks. I didn't check the last commit in the rpi-update repo, which had an obvious enough commit message. ;)

closed 2019-06-25 by TauPan