Hexxeh/rpi-firmware #205: update to 4.19.58-v7+ disables picamera

lrnzo, 2019-07-22:
hello there,

right after i updated (and rebooted) from 4.19.57-v7+ to 4.9.58.v7+ the camera modul is not available any longer. Until this update i had my youtube live stream up and running. 

now it says:

raspistill -o test.jpeg 
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

feel free to checkout the output of raspinfo: https://pastebin.com/6vt5TKsW

pelwell, 2019-07-22:
It would help enormously if you can identify the precise update that broke the camera for you. Look at this page: https://github.com/Hexxeh/rpi-firmware/commits/master
On the right of each line you see an 8-digit hexadecimal number. Running `sudo rpi-update` will install the top commit/update. For commits other than the first, include the hexadecimal number in the command, e.g. `sudo rpi-update 43fa687c`. An efficient strategy would be to confirm that the first 4.19.57 update still works and that the top commit fails, then choose one halfway between those limits. If it works then narrow the search to the upper half, otherwise look at the lower. Repeat until you have found two neighbours, one working and one broken - report the broken one.

lrnzo, 2019-07-22:
meanwhile i went back to commit ce9a76 and still have not found the camera work again. the idea came to me that i could disabe the camera in raspi-config, reboot and enable it again. sadly the pi does not come up again. i will further investigate this tomorrow.

lrnzo, 2019-07-23:
i must have made a mistake. after a fresh install of the raspbian image of 20190710 followed by apt-get update to the latest kernel foo (4.19.58-v7+) i find the camera working and my live stream from the beehive is happy again ;)

closed 2019-07-23 by lrnzo

pelwell, 2019-07-23:
No problem - come back if you do identify a real issue.