Hexxeh/rpi-firmware #207: Kernel 4.19.60 has Error when getting rpi-source

eslei322, 2019-07-29:
pi@raspberrypi:~ $ cat /proc/version
Linux version 4.19.60-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1247 SMP Thu Jul 25 14:41:19 BST 2019

pi@raspberrypi:~ $ sudo rpi-source -d /

 *** Updating rpi-source
--2019-07-29 22:33:30--  https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.164.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.164.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 11935 (12K) [text/plain]
Saving to: ‘/usr/bin/rpi-source’

/usr/bin/rpi-source 100%[===================>]  11.66K  --.-KB/s    in 0.001s

2019-07-29 22:33:31 (10.1 MB/s) - ‘/usr/bin/rpi-source’ saved [11935/11935]

 *** Set update tag: 9eaf427fc6c1a2f061d66c0ebe9f23051d49479a

 *** Restarting rpi-source

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: ce2a9f85a6fd88f8c42ef54b7bad99b42e76e403

 *** Linux source commit: 20565c94ca877a099a3c9c5fa39a0380f3d16491

 *** Download kernel source
--2019-07-29 22:33:33--  https://github.com/raspberrypi/linux/archive/20565c94ca877a099a3c9c5fa39a0380f3d16491.tar.gz
Resolving github.com (github.com)... 13.237.44.5
Connecting to github.com (github.com)|13.237.44.5|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/20565c94ca877a099a3c9c5fa39a0380f3d16491 [following]
--2019-07-29 22:33:34--  https://codeload.github.com/raspberrypi/linux/tar.gz/20565c94ca877a099a3c9c5fa39a0380f3d16491
Resolving codeload.github.com (codeload.github.com)... 192.30.255.121
Connecting to codeload.github.com (codeload.github.com)|192.30.255.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/linux-20565c94ca877a099a3c9c5fa39a0380f3d16491.tar.gz’

/linux-20565c94ca87     [              <=>   ] 155.67M  2.98MB/s    in 78s

2019-07-29 22:34:53 (1.99 MB/s) - ‘/linux-20565c94ca877a099a3c9c5fa39a0380f3d16491.tar.gz’ saved [163229688]

 *** Unpack kernel source
.............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** Add '+' to kernel release string

 *** Create symlink: /linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
/bin/sh: 1: bison: not found
make[2]: *** [scripts/Makefile.lib:196: scripts/kconfig/zconf.tab.c] Error 127
make[1]: *** [Makefile:532: syncconfig] Error 2
Makefile:619: include/config/auto.conf.cmd: No such file or directory
make: *** [Makefile:631: include/config/auto.conf.cmd] Error 2
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 359, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "/usr/bin/rpi-source", line 84, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /linux && make modules_prepare ' returned non-zero exit status 2

pelwell, 2019-07-30:
Any thoughts, @notro?

notro, 2019-07-30:
This seems to be the problem:
```
/bin/sh: 1: bison: not found
```
A search tells me that flex and bison are required since 4.16 to build the kernel.
https://www.kernel.org/doc/html/v4.19/process/changes.html#current-minimal-requirements

pelwell, 2019-07-30:
Yes, that's true (and is obvious when not reading the logs on a phone before coffee) - the first step in our kernel building instructions (https://www.raspberrypi.org/documentation/linux/kernel/building.md) is:
```
sudo apt-get install git bc bison flex libssl-dev
```
It's worth at least a note on the rpi-source Wiki.

notro, 2019-07-30:
I've added that line to the wiki.
I assumed that rpi-source would fall out of use when we got a kernel header package, but not so. I guess the Pi community is so big that the corner use cases turns out to be rather big corners.

eslei322, 2019-07-31:
thanks everyone for replying and the tip, i got the rpi-source now.

try to compile this:

https://github.com/VinnyLorrin/ES9038Q2M-Linux_Driver

on 2018-06-27-raspbian-stretch-lite with kernel 4.19.60 but got error, same code is ok on kernel 4.14.92.

*************************************************************************************************************
sudo make
make -C /lib/modules/4.19.60-v7+/source SUBDIRS=/code KBUILD_VERBOSE=0 modules
make[1]: Entering directory '/linux-20565c94ca877a099a3c9c5fa39a0380f3d16491'
  CC [M]  /code/hifibunny-codec.o
/code/hifibunny-codec.c: In function ‘hifibunny_codec_dai_startup_slave’:
/code/hifibunny-codec.c:181:35: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  struct snd_soc_codec *codec = dai->codec;
                                   ^
In file included from ./include/linux/i2c.h:30:0,
                 from /code/hifibunny-codec.c:21:
/code/hifibunny-codec.c:187:16: error: dereferencing pointer to incomplete type
   dev_err(codec->dev, "Failed to setup rates constraints: %d\n", ret);
                ^
./include/linux/device.h:1442:11: note: in definition of macro ‘dev_err’
  _dev_err(dev, dev_fmt(fmt), ##__VA_ARGS__)
           ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_dai_startup’:
/code/hifibunny-codec.c:196:41: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  struct snd_soc_codec      * codec = dai->codec;
                                         ^
/code/hifibunny-codec.c:198:6: error: implicit declaration of function ‘snd_soc_codec_get_drvdata’ [-Werror=implicit-function-declaration]
      = snd_soc_codec_get_drvdata(codec);
      ^
/code/hifibunny-codec.c:198:8: warning: initialization makes pointer from integer without a cast
      = snd_soc_codec_get_drvdata(codec);
        ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_hw_params’:
/code/hifibunny-codec.c:213:35: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  struct snd_soc_codec *codec = dai->codec;
                                   ^
/code/hifibunny-codec.c:215:2: error: implicit declaration of function ‘snd_soc_read’ [-Werror=implicit-function-declaration]
  uint8_t iface = snd_soc_read(codec, INPUT_CONFIG) & 0x3f;
  ^
/code/hifibunny-codec.c:229:2: error: implicit declaration of function ‘snd_soc_write’ [-Werror=implicit-function-declaration]
  snd_soc_write(codec, INPUT_CONFIG, iface);
  ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_set_fmt’:
/code/hifibunny-codec.c:259:40: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  struct snd_soc_codec      *codec = dai->codec;
                                        ^
/code/hifibunny-codec.c:261:8: warning: initialization makes pointer from integer without a cast
      = snd_soc_codec_get_drvdata(codec);
        ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_dac_mute’:
/code/hifibunny-codec.c:287:35: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  uint8_t genSet = snd_soc_read(dai->codec, GENERAL_SET);
                                   ^
/code/hifibunny-codec.c:290:20: error: ‘struct snd_soc_dai’ has no member named ‘codec’
   snd_soc_write(dai->codec, GENERAL_SET, genSet | 0x03);
                    ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_dac_unmute’:
/code/hifibunny-codec.c:297:35: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  uint8_t genSet = snd_soc_read(dai->codec, GENERAL_SET);
                                   ^
/code/hifibunny-codec.c:298:19: error: ‘struct snd_soc_dai’ has no member named ‘codec’
  snd_soc_write(dai->codec, GENERAL_SET, genSet & 0xFC);
                   ^
/code/hifibunny-codec.c: At top level:
/code/hifibunny-codec.c:357:15: error: variable ‘hifibunny_codec_codec_driver’ has initializer but incomplete type
 static struct snd_soc_codec_driver hifibunny_codec_codec_driver = {
               ^
/code/hifibunny-codec.c:358:2: error: unknown field ‘component_driver’ specified in initializer
  .component_driver = {
  ^
/code/hifibunny-codec.c:358:2: error: extra brace group at end of initializer
/code/hifibunny-codec.c:358:2: error: (near initialization for ‘hifibunny_codec_codec_driver’)
/code/hifibunny-codec.c:361:2: warning: excess elements in struct initializer
  }
  ^
/code/hifibunny-codec.c:361:2: warning: (near initialization for ‘hifibunny_codec_codec_driver’)
/code/hifibunny-codec.c: In function ‘hifibunny_codec_probe’:
/code/hifibunny-codec.c:395:2: error: implicit declaration of function ‘snd_soc_register_codec’ [-Werror=implicit-function-declaration]
  ret = snd_soc_register_codec(dev,
  ^
/code/hifibunny-codec.c: In function ‘hifibunny_codec_remove’:
/code/hifibunny-codec.c:406:2: error: implicit declaration of function ‘snd_soc_unregister_codec’ [-Werror=implicit-function-declaration]
  snd_soc_unregister_codec(dev);
  ^
cc1: some warnings being treated as errors
scripts/Makefile.build:309: recipe for target '/code/hifibunny-codec.o' failed
make[2]: *** [/code/hifibunny-codec.o] Error 1
Makefile:1517: recipe for target '_module_/code' failed
make[1]: *** [_module_/code] Error 2
make[1]: Leaving directory '/linux-20565c94ca877a099a3c9c5fa39a0380f3d16491'
Makefile:11: recipe for target 'all' failed
make: *** [all] Error 2

*************************************************************************************************************

try 2019-07-10-raspbian-buster-lite with kernel 4.19.60, got same error.	

any comments, suggestions, and solutions are appreciated.

pelwell, 2019-07-31:
In one of the fairly recent kernel versions (don't ask me which...) a structure field was renamed from "codec" to "component". It looks like the out-of-tree module you have was written against the older version of the ALSA API.

eslei322, 2019-08-18:
@pelwell 

thank you very much for your help.

i am an old man and my computer knowledge is too much out of date, that is why it took me so long to get the result. i definitely wouldn't get that without your guidance.

cheers

closed 2019-09-03 by eslei322