Hexxeh/rpi-firmware #208: Buffer/Memory Leak USB RTL2838 SDR Dongle

linuxonlinehelp, 2019-10-12:
Tested pi3 + pi4-4GB
Stretch ok Kernel 4.14-98
Buster Kernel 4.19-75/79 fails with Buffering Timouts over USB pulling Date from a RTL-SDR Stick V3
Package list of Test Systems for RTL-SDR Server:
https://drive.google.com/drive/folders/1gXQCKo8ImFb7i89sbFfzamYV5VKIyLJR
Devices:
Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

How to fix?? Openwebrx pulls Audio Stream over 25Seconds from the Stick
Memory set to 4096+1024 same issues, no Kernel Messages..no Errors

linuxonlinehelp, 2019-12-16:
fixed by fftw3  3.3.5 Downgrade !!

popcornmix, 2019-12-16:
This repo is only used for hosting firmware/kernel files.
Firmware/kernel bug reports should go to https://github.com/raspberrypi/linux/issues or https://github.com/raspberrypi/firmware/issues

But this doesn't sound like either of those, but a problem with a userspace library.
Try reporting it here: https://github.com/FFTW/fftw3/issues

closed 2019-12-16 by popcornmix