Hexxeh/rpi-firmware #209: How to get correct module_layout value in compiled driver

moodeaudio, 2019-10-15:
Hi,

I'm using a Pi-4 to compile a kernel7 kernel using essentially the Foundation build recipe https://www.raspberrypi.org/documentation/linux/kernel/building.md, but just for the purpose of compiling a few device drivers that are to be installed in specific versions of stock rpi-update kernels.

The problem I'm having is that the value of module_layout is different in my compiled driver vs the value in Module7.symvers for specific rpi-update kernel. This causes driver to fail to load.

For example:

4.19.75-v7+ Module7.symvers - module_layout = 0xf230cadf
My compiled driver has module_layout = 0xae9bea50

Error on reboot =

pi@rp7:~ $ dmesg | grep ax88
[    6.235563] ax88179_178a: disagrees about version of symbol module_layout

Additional:

Pi-4 is running rpi-update kernel 4.19.75
```
echo "y" | sudo PRUNE_MODULES=1 rpi-update d9321aceacfc6619b4238c6c764203b1122f2f9b
```
I'm using 4.19.75 sources for the kernel7 compile
```
https://github.com/raspberrypi/linux/commit/6d8bf28fa4b1ca0a35c0cd1dcb267fb216daf720`
```
Looking for guidance on how to get driver compiled with correct module_layout value.

-Tim

popcornmix, 2019-10-15:
This isn't the right place for questions (this repo just just hosts binaries for rpi-update).
The forum would be a better place to ask this.

Are you using Modules.symvers? (see [here](https://www.raspberrypi.org/forums/viewtopic.php?t=17666#p179845))

closed 2019-10-15 by popcornmix

moodeaudio, 2019-10-15:
The Forum was my first stop but the thread seems to have stalled...
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=252786&p=1542805#p1542676

Yes, using Module.symvers but not the same procedure as in your link. I'm just using the Foundation build recipe and then copying Module7.symvers from your repo to /usr/src/linux/Module.symvers, then  make -j4 zImage modules dtbs.

I noticed though that at the end of make modules it seems that Module.symvers file gets rewritten.

I'll have a look at the procedure in your link, thanks :-)