Hexxeh/rpi-firmware #213: RPI4 Broken PXE boot - stuck after fixup4.dat

voldermord, 2020-01-22:
Using start4.elf and fixup4.dat from any commit at and above https://github.com/Hexxeh/rpi-firmware/commit/f0fdb006abe402210fba5873c4328f9839c8480f will cause PXE netboot to hang after loading fixup4.dat from TFTP. Wireshark shows no further response after this. 

Replacing with files from previous commits allows the boot to continue without issues.
[image]

timg236, 2020-01-22:
Please can you try over_voltage=1 or over_voltage=2. There haven't been any changes to the network code in start.elf so I suspect this is related to DVFS

popcornmix, 2020-01-22:
Also could you try test firmware from here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649#p1599457

voldermord, 2020-01-23:
I've tried over_voltage=1/over_voltage=2 and the firmware at the link above. Both do not continue the pxe boot after loading fixup4.dat.

timg236, 2020-01-23:
We use network boot in the CI system so we can confirm that it is working in that firmware version. Some suggestions

* Disable any over-clocking in config.txt
* Upgrade bootloader to the latest beta
* If you have a USB serial cable then enabling BOOT_UART in the bootloader and uart_2ndstage=1 will indicate whether start.elf is running or has crashed.
* Delete fixup4.dat. A mismatch with start4.elf could cause problems but that's pretty unlikely.

Do you get the rainbow splash-screen, it might take a minute or two and what's the status of the activity (green) LED ?

voldermord, 2020-01-23:
- Stock / empty config.txt -> not working
- Bootloader: Jan 17 13:44:13 UTC 2020 (1579268653), VL805 000137ad -> not working
- Deleted fixup4.dat -> not working (works for commits before https://github.com/Hexxeh/rpi-firmware/commit/8bb38ef0267ab09d5ff6f2c3b48fff44cb69ade8
- No output on monitor, green activity light is constantly dimmed once stuck at loading start4.elf and fixup4.dat from tftp

I'd have to go out and get a usb serial cable to see what else is going on..

pelwell, 2020-01-23:
> I'd have to go out and get a usb serial cable to see what else is going on.

Alternatively, if you have a second Pi and some patch cables you could set it up as a UART receiver:

1. Make sure `enable_uart=1` is in `config.txt`.
2. If `console=serial0,115200` is in `cmdline.txt`, change it to `xconsole=serial0,115200` or delete it.
3. Connect the patch cables between the Pis:
    * pin 6 to pin 6
    * pin 8 to pin 10
    * pin 10 to pin 8
4. On the receiving Pi, run:
```
$ stty -F /dev/serial0 115200 clocal -crtscts
$ cat /dev/ttyAMA0
```
5. Boot the Pi under test.

timg236, 2020-01-24:
We found the bug in the firmware + CI test. There's a firmware fix in the pipeline. We'll update this bug once rpi-update is ready.

voldermord, 2020-01-24:
> We found the bug in the firmware + CI test. There's a firmware fix in the pipeline. We'll update this bug once rpi-update is ready.

Thats great! 
I just go some patch cables and could only see the following from another pi receiver:

`GENET STOP: 0
Starting start4.elf @ 0xcec00200`

timg236, 2020-01-24:
Yep, that's the issue. The firmware gets the wrong ethernet state and gets stuck

voldermord, 2020-02-11:
Confirmed the recent commit fixes this issue.

closed 2020-02-11 by voldermord