Hexxeh/rpi-firmware #218: PortAudio funktioniert ab Kernel 4.19.37 nicht mehr

rpipa, 2020-03-21:
Starting with kernel version 4.19.37, the following error message appears with the latest version of PortAudio:

`Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924`

1. why does the access work properly until kernel 4.19.36 and not after 4.19.37?

2. how can this error be solved?

pelwell, 2020-03-21:
Please open this issue in the kernel repo, where it belongs.

pelwell, 2020-03-21:
https://github.com/raspberrypi/linux/issues/

rpipa, 2020-03-21:
Thanks for the advice. I have attached the bug to an already existing issue.

https://github.com/raspberrypi/linux/issues/3077

closed 2020-03-21 by rpipa