Hexxeh/rpi-firmware #219: Kernel source 4.19.108+

AMuszkat, 2020-04-08:
Hi,

We are trying to compile the ma120x0p codec driver for the 4.19.108+ release but we still have symbols issues. do you have an exact link to copy the kernel source ?

Currently we are trying [this one](https://github.com/raspberrypi/linux/tree/2fab54c74bf956951e61c6d4fe473995e8d07010) but does not work.

Best,
Ariel

pelwell, 2020-04-08:
That's a known problem with the Module.symvers files that should be solved in the next release. In the meantime, see this post for a temporary workaround: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=269769&start=50#p1639775

MrEngman, 2020-04-08:
Hi pelwell,

The issue I found trying to compile out of tree wifi drivers was only with the 5.4 kernel in raspberry/firmware BRANCH=next.

I have never found an issue with the Module.symvers file previously and I have literally compiled thousands of driver files previously for kernel versions from kernel 3.xx to the current 4.19.114. Essentially every time there is an update to the kernel available from Hexxeh/rpi-update

I am not sure this fix I found, which your link points to, is valid but it has temporarily fixed the problem I found.

MrEngman

pelwell, 2020-04-08:
@AMuszkat Rebuilding the whole kernel doesn't take that long on a Pi 4 - that way you'll avoid any symbol incompatibility.

AMuszkat, 2020-04-08:
Thanks @pelwell . This is just a specific case we need for moode audio player, thats why we are working with 108+ as the next release will be based on that Kernel and it is already compiled and almost finished. Rebuilding the whole kernel with the images (kernel.img, kernel7.img, etc...) works fine.

Have tried recompiling with the gcc 5 plugin and it does not work.

pelwell, 2020-04-08:
You appear to have the correct commit for the 108+ build (2fab54c74bf956951e61c6d4fe473995e8d07010), so there must be some other reason for the incompatibility. What process are you following to build just that one codec driver?

AMuszkat, 2020-04-08:
I'm cross compiling according to the [pi guide](https://www.raspberrypi.org/documentation/linux/kernel/building.md). Then transfer the snd-soc-ma120x0p.ko to the pi and run sudo depmod 4.19.108+. The pi is running [this firmware](https://www.raspberrypi.org/documentation/linux/kernel/building.md).