Hexxeh/rpi-firmware #222: PoE HAT Fan 100% in Kernel Version 4.19.118

pthoelken, 2020-04-28:
Hello Community, 

in the last release 4.19.118 the PoE HAT Fan on RPI4B (4GB) is going to 100% instant and all the time. 

The params in the /boot/config.txt already set it up but nothing helps.

[image]

pelwell, 2020-04-28:
We recently added two more threshold values in the overlay. The defaults are now:
```
Name:   rpi-poe
Info:   Raspberry Pi PoE HAT fan
Load:   dtoverlay=rpi-poe,<param>[=<val>]
Params: poe_fan_temp0           Temperature (in millicelcius) at which the fan
                                turns on (default 40000)
        poe_fan_temp0_hyst      Temperature delta (in millicelcius) at which
                                the fan turns off (default 2000)
        poe_fan_temp1           Temperature (in millicelcius) at which the fan
                                speeds up (default 45000)
        poe_fan_temp1_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp2           Temperature (in millicelcius) at which the fan
                                speeds up (default 50000)
        poe_fan_temp2_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 2000)
        poe_fan_temp3           Temperature (in millicelcius) at which the fan
                                speeds up (default 55000)
        poe_fan_temp3_hyst      Temperature delta (in millicelcius) at which
                                the fan slows down (default 5000)
```
Try specifying all 4 operating points. If you want to raise the point at which the fan cuts in, but by default it should come on slow at 40C.

pthoelken, 2020-04-28:
So I will try it as you can see: 

`dtparam=poe_fan_temp0=60000,poe_fan_temp0_hyst=2000
dtparam=poe_fan_temp1=63000,poe_fan_temp1_hyst=3000
dtparam=poe_fan_temp2=65000,poe_fan_temp2_hyst=2000
dtparam=poe_fan_temp3=67000,poe_fan_temp3_hyst=2000`

pthoelken, 2020-04-28:
Perfect. Solved.

closed 2020-04-28 by pthoelken