Hexxeh/rpi-firmware #224: No Audiodevice found since Commitnumber 4e0af7eb74aad67ad0e219f9ec5e975e84362100

Retro1968, 2020-05-13:
Raspberry Pi 4 with RetroPie 4.6 and Raspbian Buster.

Since Commitnumber 4e0af7eb74aad67ad0e219f9ec5e975e84362100 Raspbian Buster do not find an Audiodevice anymore. 

Restoring the Commitnumber 8d64ec7016906fb1f2bbe937dd7401705c5acf33 solves that issue, but it would be kind if you fix that issue in an upcomming firmware.

Cheers
Retro1968

pelwell, 2020-05-13:
That commit introduces Device Tree changes that pass additional command line parameters to the kernel in order to enable the new discrete audio devices in ALSA. The audio plugin in Buster is too old to know about the split devices.

We don't guarantee that old OSs will continue to work with new kernels - at some point you need to update to Stretch or stop updating your kernel. However, on this occasion you can revert to the old behaviour by adding `snd_bcm2835.enable_compat_alsa=1 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_headphones=0` to your cmdline.txt.

Retro1968, 2020-05-13:
Danke für die Info.

closed 2020-05-13 by Retro1968