Hexxeh/rpi-firmware #225: Kernel sources for corresponding rpi-update kernel release

moodeaudio, 2020-06-07:
Hi,

How to  determine which commit in the 5.4.y raspberry/linux source tree corresponds to a given rpi-update kernel release? For example which commit would be used to download the sources for the rpi-update 5.4.44 kernel?

In the 4.19 branch there used to be two commits after each rpi-update kernel release (example below) but these don't appear in the 5.4.y branch.

https://github.com/raspberrypi/linux/commits/rpi-4.19.y
Merge remote-tracking branch 'stable/linux-4.19.y' into rpi-4.19.y
Linux 4.19.126

popcornmix, 2020-06-07:
The merge commits in 4.19 don't correspond with rpi-update releases.
They are just the point where we have merged in latest stable tree.
Use the git_hash file to find the commit in the raspberry/linux tree the release was built from.
e.g.
```
$ cat /boot/.firmware_revision
98d0973018dd2374ede285791285e352a5060d55
```
Then look at that revision of git_hash:
```
$ curl https://raw.githubusercontent.com/Hexxeh/rpi-firmware/65ab2e03576e6653bb45f5e68b80268a4401eecf/git_hash
3f54521ea7236a4e70dcb12ebc76ecafd54b8858
```
and use that to find source in the linux tree:
https://github.com/raspberrypi/linux/tree/3f54521ea7236a4e70dcb12ebc76ecafd54b8858

moodeaudio, 2020-06-07:
Perfect, thanks :-)

closed 2020-06-07 by moodeaudio