Hexxeh/rpi-firmware #226: 5.4.44 an KPPS?

by, 2020-06-07:
When trying kernel PPS w/ ntpsec, the current 64-bit kernel 5.4.44 w/ 32-bit userland throws a warning on a Pi4:
`REFCLOCK: refclock_params: kernel PLL (hardpps, RFC 1589) not implemented`.
Any chance to get this included in the regular build? – Thank you!

pelwell, 2020-06-07:
Is this a feature that you need, or are you just concerned about the warning?

by, 2020-06-07:
I need it for my local NTP servers, they are much more precise with it, thank you!

pelwell, 2020-06-07:
Looking at the Kconfig files, this seems to correspond to option CONFIG_NTP_PPS:
```
config NTP_PPS
	bool "PPS kernel consumer support"
	depends on !NO_HZ_COMMON
	help
	  This option adds support for direct in-kernel time
	  synchronization using an external PPS signal.

	  It doesn't work on tickless systems at the moment.
```
Unfortunately Pi kernels _are_ tickless, so CONFIG_NO_HZ_COMMON=y and NTP_PPS can't be selected.

by, 2020-06-07:
Guess it relates to at least this old request https://github.com/raspberrypi/linux/issues/1218
Would be great to be able to use kernel for PPS when a GPS HAT is attached and it is run as an NTP server...

pelwell, 2020-06-08:
Yes - that's exactly the same issue. We're not going to change our scheduling model just for this use case, so you have three option:
1. Persuade somebody to do the work to make PPS work in tickless mode.
2. Build your own kernel with "Tickless System (Dynamic Ticks)" disabled and the PPS support enabled. There are detailed build instructions available at https://www.raspberrypi.org/documentation/linux/kernel/building.md - it's probably less than an hour on Pi 4, and much quicker than that for subsequent updates.
3. Decide that the current accuracy is good enough.

by, 2020-06-08:
Thanks; while I'm hoping for (1), I guess I'll need to try (2) with (3) as my fallback...

Ruffio, 2020-08-26:
Can this be closed?

by, 2022-02-12:
Solved via https://www-kiokoman-eu-org.translate.goog/index.php/8-default/31-howto-raspberry-pi-4-build-kernel-for-ntp?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=de

closed 2022-02-12 by by