Hexxeh/rpi-firmware #227: 5.4.47 breaks bluetooth on rpi4

knro, 2020-06-18:
After updating to 5.4.47, bluetooth fails to initialize on Raspberry PI 4 (4GB).

hciuart shows this:

```
Jun 18 10:06:50 stellarmate systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Jun 18 10:07:20 stellarmate btuart[387]: Initialization timed out.
Jun 18 10:07:20 stellarmate btuart[387]: bcm43xx_init

```

has to revert to a50c7d5eebb351d16665eabcedad992cdc167537 to make it work again.

pelwell, 2020-06-18:
Thanks for the report - this overlaps with a problem I've been looking at.

pelwell, 2020-06-18:
Can you use the "SKIP_KERNEL=1" option to see if it was the firmware update or the kernel update that broke it?

pelwell, 2020-06-18:
I can answer that - an rpi-update followed by a "SKIP_KERNEL=1 rpi-update a50c7d5" leaves you with working Bluetooth, so the breakage is the result of a firmware change.

pelwell, 2020-06-18:
The Bluetooth issue is understood and will be fixed in the next release. Could you please create a new issue for the lack of headphone audio.

pelwell, 2020-06-22:
The latest rpi-update release (https://github.com/Hexxeh/rpi-firmware/commit/634e380a4d041492f859712bd2c81112a535b515) includes the fix for the Bluetooth regression.

closed 2020-07-24 by knro