Hexxeh/rpi-firmware #228: No audio from 3.5mm jack on 5.4.45-v7l+

mrgfisher, 2020-06-18:
I have no audio from the 3.5mm jack, I *think* it was working on 5.4.44-v8+. I actually updated to latest (rpi-update and rpi-eeprom-update), then reverted back to the previous commit of 
a50c7d5  I do this out of curiosity and am aware things may change and break, so not complaining :-)

Its a 8G Pi4 on regular 32-bit rasbian, fully uptodate via apt. It boots entirely from SSD, though that may not be pertitent.

Using the desktop I've tried various things, but can only get audio out by selecting HDMI.

If you have some commands to execute or log files to provide then please let me know how I can help.

Thanks!

popcornmix, 2020-06-18:
Can you revert back to 5.4.44 and confirm that worked? I can't think of a reason why it would have stopped working between those two points, but if you can confirm it we can look more closely.

mrgfisher, 2020-06-18:
Still no audio.

rpi-udpate 65ab2e0  - and uname -a now reports 5.4.44-v7l+
checked nothing forced in config.txt (though it was working before I started trying updates!)
checked the output is 'analog' in the UI (I'm sure its swapping between names too, but could easily be going mad!)

Audio is still working over HDMI, and irrespective of settings, it still comes out of HDMI (this may be correct behaviour, never plugged speakers into hdmi montior before today, always used the 3.5mm jack on the board).

Again, many thanks for taking a look, if it 'works for you' then just close this ticket as it must be something odd in my setup. I'll also attach the config.txt file if that is of any help (I overclock to 2000, but have been done this for months on my Pi4s).

[config.txt](https://github.com/Hexxeh/rpi-firmware/files/4800331/config.txt)

popcornmix, 2020-06-18:
https://github.com/raspberrypi/linux/commit/0339fab951121e5ff8c85791e3c99ee1b98b3d35#diff-fb8cf9a1a0b868557516caf54f4e966b did affect alsa behaviour.

The raspberry pi os gui audio selector should handle the new configuration.
But you could try undoing this change if it used to work.
I believe adding to cmdline.txt
```
snd_bcm2835.enable_compat_alsa=1 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_headphones=0```

will give the old behaviour.

mrgfisher, 2020-06-18:
No luck - with the amended cmdline.txt (double checked that I've copy and pasted correct), checked in raspi-config for anything forced etc. The odd UI behavior is that when I switch to analog it reset the volume to 2, when I increase it, it then looses the green tick to indicate that the UI panel item is set to Analog as opposed to HDMI.

In dmesg, I can see:
[    5.386439] brcmfmac: F1 signature read @0x18000000=0x15264345
**[    5.399754] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6**
[    5.401243] usbcore: registered new interface driver brcmfmac
[    5.412896] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
**[    5.635262] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.649027] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2**
[    6.596583] random: crng init done

I've used bold for the items in red...

Have switched back to HDMI and again irrespective of the above tweaks it works.

If audio works for you, then must just be my system and the various updates its been through... so no worries.

DonFlymoor, 2020-06-18:
I can confirm this bug. I only use the headphone jack, and ALSA stopped working completely after running rpi-update, at least in regards to the headphone jack.

popcornmix, 2020-06-19:
Are you on raspbian buster? Is it updated?

mrgfisher, 2020-06-19:
Yep:
pi@attea:~ $ uname -a
Linux attea 5.4.45-v7l+ #1321 SMP Wed Jun 10 17:39:20 BST 2020 armv7l GNU/Linux
pi@attea:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@attea:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Fetched 13.0 MB in 6s (2,326 kB/s)   
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
pi@attea:~ $

popcornmix, 2020-06-19:
I've just tried on an updated system
```
pi@pi4:~ $ uname -a
Linux domnfs 5.4.47-v7l+ #1322 SMP Wed Jun 17 17:58:52 BST 2020 armv7l GNU/Linux
```
I booted to desktop. I used right-click on speaker icon and chose "AV Jack".
I launched neverball and heard game audio from 3.5mm jack.
I quit neverball and used right-click on speaker icon and chose "HDMI"
I launched neverball and heard game audio from HDMI.
and repeated and that worked as expected.

Note: switching audio wile neverball was running had no immediate effect. It only had an effect the next time it restarted.

Can I also confirm you are using the default fkms driver, rather than the experimental kms driver?
Can you post output of "aplay -L"

mrgfisher, 2020-06-19:
Hmm, odd - my options from right click on the speaker icon are Analog and HDMI.

aplay -L reports:

pi@attea:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
output
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dmix:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=2
    bcm2835 ALSA, bcm2835 IEC958/HDMI1
    Hardware device with all software conversions
usbstream:CARD=ALSA
    bcm2835 ALSA
    USB Stream Output
pi@attea:~ $ 

Thanks again @popcornmix

popcornmix, 2020-06-19:
Okay, you've installed pulseaudio - something we don't use by default in Raspberry Pi OS.
I'd suggest removing that unless you have a compelling need for it.

You should then see:
```
pi@pi4:~/projects/alsa-lib $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=b1
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Default Audio Device
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
```
and I suspect analogue audio will start working.

mrgfisher, 2020-06-19:
My Pi makes sounds again! :-) 

I did sudo apt autoremove pulseaudio 
plus a reboot. It's not tidied up the output of aplay -L much, but now it does work. The output (right click on speaker) still says Analog rather than AV Jack, but no worries.

Many thanks @popcornmix and sorry to have reported a software config problem here. I'll close this issue unless you want me to do any futher checks...

closed 2020-06-19 by mrgfisher