Hexxeh/rpi-firmware #230: ASoC: failed to init link HiFiBerry Digi+ Pro: -517

hellgelino, 2020-06-28:
Hi,

i cant use my hifiberry digi+ pro card anymore. Im using the newest 64 bit Kernel on my pi 4 8GB.
ASoC: failed to init link HiFiBerry Digi+ Pro: -517

OS Basic Image max2play

Greats
Helge

pelwell, 2020-06-28:
What is the most recent kernel that does give you a working HiFiBerry?

The error number -517 is -EPROBE_DEFER - it means that some dependency isn't available yet and the kernel will try again later. This isn't usually an error - normally drivers don't add a message to the kernel log for this error number - but if the required dependency is never met then that is an error.

Could you post:
1. the complete output of `dmesg`
2. the output of `lsmod`

Ruffio, 2020-08-26:
@hellgelino is this still an issue or has it been resolved? Can you provide the above information if still an issue?

closed 2020-08-26 by hellgelino