Hexxeh/rpi-firmware #231: Wiki page said,

PaS3, 2020-07-13:
Wiki page said,

> Options
> 
> If you'd like to set a different GPU/ARM memory split, then define gpu_mem in /boot/config.txt.
> 
> To upgrade/downgrade to a specific firmware revision, specify its Git hash (from the https://github.com/Hexxeh/rpi-firmware repository) as follows:
> 
> sudo rpi-update fab7796df0cf29f9563b507a59ce5b17d93e0390
> 

 but there is no list of possible hashes      ?    

 
I could generate something like a sha1sum git commit contents with date/version history ( echo "$(cat /boot/kernel7.img) $(date)"  | sha1sum )  
but not sure what is the date and version history. 

Please help.

pelwell, 2020-07-13:
The hashes are on this page: https://github.com/Hexxeh/rpi-firmware/commits/master

Note that you may need to enable Desktop mode if you are using a mobile browser.

PaS3, 2020-07-13:
Thanks, it may have been my dns configs.