Hexxeh/rpi-firmware #233: Update to 5.4 branch in raspberrypi-kernel package has broken Kodi

eccgecko, 2020-07-20:
Since running apt update and upgrade (and then full-upgrade as per a recommendation), I am now on the 5.4 branch of the kernel on my 32-bit 3B+. To be exact, I am on 5.4.51-v7+.
As far as I can tell, this was the only thing on my Pi that was upgraded today (as in, the kernel, firmware, and bootloader etc, and not any other software packages), and so this is the only thing that has changed.

Since this update, the Kodi I have installed on my system is broken. I use my system primarily as a secondary desktop machine in the day, with occasional use of Kodi in the evenings for proper YouTube and Netflix playback etc. To get smooth Netflix playback, my Pi was overclocked, and on the 4.19 branch my overclock settings were working perfectly. Had a very stable system, with smooth playback of Netflix.

Since the update to 5.4, Kodi was no longer starting up properly. It was very hit-and-miss as to whether it would start up at all, with it often freezing on the splash screen. If it did get to the Kodi home screen, the screen would tear, and if it did not tear, I could occasionally browse through a few menus before it would freeze. The strange thing is, it was not crashing, only freezing. The pi itself was still running fine.

To troubleshoot, I disabled all non-standard settings in config, and so I now _think_ I have narrowed this down to the gpu_freq overclock in my config.txt file. I disabled all my overclock settings, and enabled them one by one, each time restarting and opening Kodi to see if the issue persisted. It was with the gpu_freq=500 that the issue was coming back.

Can anyone else confirm if gpu_freq=500 is causing issues with their Kodi installation since upgrading to the 5.4 branch of the raspberrypi-kernel package?

I am running the latest Kodi 18.7 package installed from the standard raspbian repo.

eccgecko, 2020-07-20:
I believe I have narrowed this down further to two of the constituent settings of gpu_freq, and I think the problematic settings are h264_freq and v3d_freq, with h264_freq being the more consistently troublesome.

I can enable all my old 4.19 overclock settings with no issues, except for h264_freq and v3d_freq.
Once I set h264_freq to `h264_freq=500` Kodi consistently freezes upon starting. Disabling all other overclock settings but leaving `h264_freq=500` results in the same behaviour. `v3d_freq=500` also causes some issues, but with less consistency than the problematic h264_freq overclock.

Like I said, before the update to 5.4 branch today, `gpu_freq=500` was working fine, as well as my other overclock options, so this is definitely an issue introduced with the 5.4 update.

I hope it's helpful that I've managed to narrow it down to these particular overclock settings. Please let me know if there are more settings or logs I should disclose.

popcornmix, 2020-07-21:
This repo is used for distributing binaries with rpi-update.
Can you report your issue here: https://github.com/raspberrypi/firmware/issues

closed 2020-07-21 by popcornmix