Hexxeh/rpi-firmware #234: cgroups not enabled?

fransking, 2020-07-23:
Details here https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=280656&p=1701228#p1701228

but in summary 5.4.51-v7+ (RPI 3B+) reports this in the syslog

Jul 23 18:27:01 pi-kub1 k3s[763]: time="2020-07-23T18:27:01.047449579+01:00" level=fatal msg="failed to find memory cgroup, you may need to add \"cgroup_memory=1 cgroup_enable=memory\" to your linux cmdline (/boot/cmdline.txt on a Raspberry Pi)"

but 5.4.51-v7l+ seems ok (RPI 4)

fransking, 2020-07-23:
I should point out that I do have cgroup_memory=1 cgroup_enable=memory added to /boot/cmdline.txt

:)

carpie, 2020-07-26:
This happened to me as well. My k3s cluster failed to operate after doing an `apt upgrade`. It appears in the v5 kernel memory cgroups aren't configured? `cat /proc/cgroups` does not show a `memory` entry.

If it helps anyone, I ran `sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` to revert to the 4.19.118 kernel to rescue my cluster.

fransking, 2020-08-02:
Looks like https://github.com/Hexxeh/rpi-firmware/commit/56f62dee2a2385fe030724f7b39344fbc8c8d7ab resolves this issue and a fresh sudo rpi-update (no hash) gives me a 5.4.51-v7+ kernel with cgroups

~ $ cat /proc/cgroups 
#subsys_name	hierarchy	num_cgroups	enabled
cpuset	6	16	1
cpu	8	47	1
cpuacct	8	47	1
blkio	4	47	1
memory	3	83	1
devices	2	47	1
freezer	9	16	1
net_cls	7	16	1
pids	5	51	1

closed 2020-08-02 by fransking