Hexxeh/rpi-firmware #239: xrandr broken: failed to get size of the gamma for the output default

alanbork, 2020-09-01:
after rpi-update 039088f7d929b0d1cc49e8d2012df0652f908518

startx works fine, but if I try to run xrandr:
xrandr: Failed to get size of the gamma for the output default
and it just reports one resolution:
576x416

as far as I know my configuration is stock other than running rpi-update. And yes, it worked fine minutes before I ran rpi-update.

alanbork, 2020-09-02:
realized I left out some useful details:

popcornmix, 2020-09-02:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

alanbork, 2020-09-02:
That would be a lot easier if there was a way to tell what the previous
commit number (git hash) I was running was. Then I could do a binary
search, rather than a serial search. I was at least a month behind, maybe
two before updating.

alanbork, 2020-09-02:
This is going to be tricky to diagnose because it seems to come and go with reboots.  Either it always works fine or it always fails.