Hexxeh/rpi-firmware #242: RetroArch FPS issues in mainmenu

lefturgen, 2020-09-29:
after updating to https://github.com/Hexxeh/rpi-firmware/commit/ca14404412d6c2a9966d7766bbad6cc24989a33f  and later, FPS in RetroArch emulator mainmenu is dropped down, menu scrolling is not smooth. Latest stable https://github.com/Hexxeh/rpi-firmware/commit/039088f7d929b0d1cc49e8d2012df0652f908518 

i tried updating with SKIP_KERNEL option but it didn't help.

popcornmix, 2020-09-29:
What model of Pi?
What was the hdmi mode before and after the firmware update?
You can use "tvservice -s" to check this.

lefturgen, 2020-09-29:
Oh, sorry, forgot about my hardware...

RPI 4B 1.1

Before:
state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive

after:
state 0xa [HDMI CUSTOM RGB lim 16:9], 3840x2160 @ 30.00Hz, progressive

thanks for the hint. hdmi_ignore_edid=0xa5000080 works for me.

Why resolution changed after the update?

popcornmix, 2020-09-29:
This will be a side effect of " firmware: hdmi/edid: Reduce the bias to all but the first detailed timing"
Ping @6by9

Ruffio, 2020-12-27:
@lefturgen is this issue still relevant? Else please close it, thanks :-)

closed 2020-12-27 by lefturgen