Hexxeh/rpi-firmware #243: ttySCx missing in /dev

Louro, 2020-10-30:
Hey

Looks like after upgrading tho 5.4.72 and newest 73 the /dev/ttySCx is missing

 uname -na                 
Linux CODESYS 5.4.73-v7+ #1360 SMP Thu Oct 29 15:53:18 GMT 2020 armv7l GNU/Linux

$ sudo cat /proc/tty/drivers
/dev/tty             /dev/tty        5       0 system:/dev/tty
/dev/console         /dev/console    5       1 system:console
/dev/ptmx            /dev/ptmx       5       2 system
/dev/vc/0            /dev/vc/0       4       0 system:vtmaster
unknown              /dev/ttySC    239 0-7 serial
acm                  /dev/ttyACM   166 0-255 serial
ttyAMA               /dev/ttyAMA   204 64-77 serial
ttyprintk            /dev/ttyprintk   5       3 console
serial               /dev/ttyS       4      64 serial
pty_slave            /dev/pts      136 0-1048575 pty:slave
pty_master           /dev/ptm      128 0-1048575 pty:master
unknown              /dev/tty        4 1-63 console

 $ ls /dev
autofs           cuse       gpiochip2  initctl  loop5         mmcblk0p1  ram1   ram4    rtc      stdout  tty15  tty23  tty31  tty4   tty48  tty56  tty7       urandom  vcs3   vcsa5     vcsu5    video16
block            disk       gpiochip3  input    loop6         mmcblk0p2  ram10  ram5    rtc0     tty     tty16  tty24  tty32  tty40  tty49  tty57  tty8       usb      vcs4   vcsa6     vcsu6    watchdog
btrfs-control    dma_heap   gpiomem    kmsg     loop7         mqueue     ram11  ram6    serial   tty0    tty17  tty25  tty33  tty41  tty5   tty58  tty9       v4l      vcs5   vcsm      vhci     watchdog0
bus              fb0        hidraw0    log      loop-control  net        ram12  ram7    serial0  tty1    tty18  tty26  tty34  tty42  tty50  tty59  ttyACM0    vchiq    vcs6   vcsm-cma  video10  zero
cachefiles       fd         hwrng      loop0    mapper        null       ram13  ram8    serial1  tty10   tty19  tty27  tty35  tty43  tty51  tty6   ttyAMA0    vcio     vcsa   vcsu      video11
char             full       i2c-0      loop1    media0        ppp        ram14  ram9    shm      tty11   tty2   tty28  tty36  tty44  tty52  tty60  ttyprintk  vc-mem   vcsa1  vcsu1     video12
char_dev         fuse       i2c-1      loop2    media1        ptmx       ram15  random  snd      tty12   tty20  tty29  tty37  tty45  tty53  tty61  ttyS0      vcs      vcsa2  vcsu2     video13
console          gpiochip0  i2c-10     loop3    mem           pts        ram2   raw     stderr   tty13   tty21  tty3   tty38  tty46  tty54  tty62  uhid       vcs1     vcsa3  vcsu3     video14
cpu_dma_latency  gpiochip1  i2c-11     loop4    mmcblk0       ram0       ram3   rfkill  stdin    tty14   tty22  tty30  tty39  tty47  tty55  tty63  uinput     vcs2     vcsa4  vcsu4     video15

dmesg | grep tty
[    0.000864] printk: console [tty0] enabled
[    2.769802] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 81, base_baud = 0) is a PL011 rev2
[    2.777032] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 53, base_baud = 62500000) is a 16550
[    9.331081] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
[   10.098718] serial serial0: tty port ttySC0 registered
[   10.099217] serial serial1: tty port ttySC1 registered

Louro, 2020-10-31:
Maybe I managed to fix it, just need to tested with RS485 bus.

in /boot/config.txt i have the wrong overlay file, was using sc16is752-spi0-ce1.dtbo and changing the config.txt to sc16is752-spi0 it shows now the serial ports

This a specific configuration for Andino IO board.

# SPI on
dtoverlay=spi0-cs,cs0_pin=8,cs1_pin=7
dtparam=spi=on

# 2. UART
# /dev/ttySC0 = RS485
# /dev/ttySC1 = RS232
dtoverlay=sc16is752-spi0,int_pin=24

pelwell, 2020-10-31:
You had me worried there for a moment, but the overlay has been adjusted to work with serdev support enabled (by moving the clock node out of the `sc16is752@0` node).

Louro, 2020-11-02:
I have tested now with RS485 and its NOT working. 

cat /proc/tty/drivers
/dev/tty             /dev/tty        5       0 system:/dev/tty
/dev/console         /dev/console    5       1 system:console
/dev/ptmx            /dev/ptmx       5       2 system
/dev/vc/0            /dev/vc/0       4       0 system:vtmaster
unknown              /dev/ttySC    239 0-7 serial
usbserial            /dev/ttyUSB   188 0-511 serial
ttyAMA               /dev/ttyAMA   204 64-77 serial
ttyprintk            /dev/ttyprintk   5       3 console
serial               /dev/ttyS       4      64 serial
pty_slave            /dev/pts      136 0-1048575 pty:slave
pty_master           /dev/ptm      128 0-1048575 pty:master
unknown              /dev/tty        4 1-63 console

stevenmacedo, 2023-02-03:
@Louro i facing the same issue, do you have solution?

thank you.