Hexxeh/rpi-firmware #245: /dev/ttyUSB3 connection problem

will-caruana, 2020-11-02:
When running Kernel: Linux 5.4.51-v7l+ I am unable to connect to my cell modem on /dev/ttyUSB3

The temporary fix I am using is running:
sudo rpi-update a995f0c9a9b3e5dbd2a41f69f5c4c82cdf33c7ec

rocketjosh, 2020-11-02:
Adding more to this issue: 

The cell modem in question is the Quectel BG96, and it appears as though some changes in 5.4.66 have affected how USB3 is treated?

From this [link](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.4.66):

    Change rules for EC21, EC25, BG96 and EG95 to match vendor specific
    serial functions only, to prevent binding to class functions. Require
    2 endpoints on ff/ff/ff functions, avoiding the 3 endpoint QMI/RMNET
    network functions.

As @will-caruana mentions, reverting to 5.4.65 allows for BG96 to load correctly.

pelwell, 2020-11-02:
If you draw the traditional 4-box diagram (showing all combinations of two binary variables) with upstream/downstream on one axis and Raspberry Pi/third-party on the other, this issue falls squarely into the least supported quadrant - it's an upstream change causing a problem with a third-party product.

If you find an upstream fix for it, or even a fix written by somebody else, there's a chance we might accept it in our Linux repo (which is where the issue should have been reported) , but we're not going to investigate the problem.

Ruffio, 2020-12-27:
@will-caruana  is this issue still relevant? Else please close it, thanks :-)