Hexxeh/rpi-firmware #246: VLC hang and need a reboot to play video

SebDominguez, 2020-11-18:
I'm trying to find out why now all my with some video playback are crashing.

I tried [there](https://github.com/RPi-Distro/vlc/issues/31) to get some help but since I had the same problem using omxplayer: maybe this is due to the video core of the RPi4...

SebDominguez, 2020-11-19:
well I didn't managed to make the 3B+ crash. I'll let the pi running over night

Ruffio, 2020-12-27:
@SebDominguez  is this issue still relevant? Else please close it, thanks :-)

SebDominguez, 2020-12-27:
I still experience deadlocks using a raspberry pi 4b and vlc or omxplayer.

I'm now using RPI3b+ for my video project now as a quickfix but I wish I could use recent hardware to future proof my stuff...

SebDominguez, 2021-05-04:
Funny enough, I've now upgraded my Octoprint server with a Raspberry Pi 4 and the video stream for long prints crash.

there is the output of the `vcgencmd version`command:

```
Apr 12 2021 15:53:21
Copyright (c) 2012 Broadcom
version 0227a0840ee8c94294cf3b2c4b2d42031a58f05a (clean) (release) (start_x)
```

I believe that I have the latest firmware installed since `sudo rpi-update` doesn't bump be to any new version.

I'm now rolling back to my old RPi3 with the latest version of Octoprint to see if the video also crash.