Hexxeh/rpi-firmware #248: br_netfilter missing in next (5.10.0-rc4)

Links2004, 2020-11-22:
I am currently trying to run kubernetes with kube-proxy on a rpi4,
but the kernel module br_netfilter is missing.

please build it again, like in 5.4.77.

Thanks

resulting errors:
```
1122 18:51:27.514843       1 network_policy_controller.go:156] Starting network policy controller full sync goroutine
E1122 18:51:27.517410       1 network_routes_controller.go:229] Failed to enable netfilter for bridge. Network policies and service proxy may not work: exit status 1
E1122 18:51:27.517686       1 network_routes_controller.go:232] Failed to enable iptables for bridge. Network policies and service proxy may not work: open /proc/sys/net/bridge/bridge-nf-call-iptables: no such file or directory
```

tests:
```
[root@rpi4]~ # modprobe br_netfilter
modprobe: FATAL: Module br_netfilter not found in directory /lib/modules/5.10.0-rc4-v7l+
[root@rpi4]~ # ls /proc/sys/net/bridge
ls: cannot access '/proc/sys/net/bridge': No such file or directory
[root@rpi4]~ # ls /proc/sys/net/
core  ipv4  ipv6  netfilter  nf_conntrack_max  unix
```

pelwell, 2020-11-23:
Config issues (and anything else kernel-related) are better reported in our kernel repo: https://github.com/raspberrypi/linux/issues/
However, since this is a clear regression caused by an upstream config change, it's now fixed in the rpi-5.10.y branch and the module will appear in future builds of 5.10.

Links2004, 2020-11-23:
thanks for the fixing and info, will keep that in mind.

may It makes sense to add the Info that kernel related issues are handled in a other git.
I found this repo url via reading the rpi-update code on the pi, but not sure how common this is :)

pelwell, 2020-11-23:
If you Google for "raspberry pi kernel issues" you will end up in the right place. Searching just for "raspberry pi kernel" takes you to the documentation for building it, the first paragraph of which links to the source repo.

Links2004, 2020-11-29:
5.10.0-rc5 Installed and k8s is working fin.

Thanks

closed 2020-11-29 by Links2004