Hexxeh/rpi-firmware #250: System freeze when play audio

matzrm, 2020-12-10:
Using lib-gstreamer to play audio with a Qt\Qml app freeze the system with kernel v79 and v81.
Note that the system freeze not only my app, for example also reboot through ssh not works. 

Meanwhile I'm sure that this commit works: 0db18b19afb84f78cdd8c708d54c597c67d03441

popcornmix, 2020-12-10:
Have you confirmed reverting just the kernel back to 5.4.59 (and not other packages) restores functionality?
It sounds like this issue may be related to the [switch to pulseaudio](https://www.raspberrypi.org/blog/new-raspberry-pi-os-release-december-2020/)

matzrm, 2020-12-10:
Yes, just reverting to 59 works.

popcornmix, 2020-12-10:
Can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this error.

matzrm, 2020-12-11:
The first version with error is ca14404412d6c2a9966d7766bbad6cc24989a33f

sairuk, 2020-12-27:
I have hard locks with audio on 453e49bdd87325369b462b40e809d5f3187df21d and above. 

I'm stuck on 5.4.81 for now. 

Can replicate by playing any video with audio through vlc or browser after upgrading.

Rolled through a few releases, (5.10.2 caused issues the other day as well)

<pre>
bump to 5.4.79  039088f7d929b0d1cc49e8d2012df0652f908518 OK
still OK        8a4d444d6a621ff56fa8941480c40fe4587c7a98 OK
bump to 5.4.81  c91312ae1df5011bd77826e49d2b914b6d0e9b08 OK
                dea7234943c604462e476a8afc13c587418e8709 OK
bump to 5.4.81  453e49bdd87325369b462b40e809d5f3187df21d BAD 5.4.83
bump to 5.10.0  7378e549a785bd486c19dc2ed737ca4a1e983351 BAD 
</pre>

let me know if i can grab some more info to help

popcornmix, 2020-12-27:
@sairuk can you try with latest rpi-update firmware? I believe that fixes an issue with audio and dual hdmi.

sairuk, 2020-12-27:
@popcornmix thanks. I updated to da59cb1161dc7c75727ec5c7636f632c52170961 (kernel v5.10.3) and unfortunately the hard lock problem persisted 
<pre>
Linux twofar 5.10.3-v7l+ #1386 SMP Sun Dec 27 14:05:40 GMT 2020 armv7l GNU/Linux
</pre>

providing I don't load `snd_bcm2835` (using rmmod, blacklist, dtparam) everything is working well; just no sound obviously but also no hard lock.

As soon as I `modprobe snd_bcm2835` and try to play the video, hard lock again

<pre>
[  265.326320] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[  265.340699] bcm2835_audio bcm2835_audio: card created with 4 channels
[  265.349276] bcm2835_audio bcm2835_audio: card created with 2 channels
[  265.353421] bcm2835_audio bcm2835_audio: card created with 2 channels
</pre>

I have rolled back to dea7234943c604462e476a8afc13c587418e8709 again for the time being
<pre>
Linux twofar 5.4.81-v7l+ #1378 SMP Mon Dec 7 18:43:09 GMT 2020 armv7l GNU/Linux
</pre>

jbsilva, 2020-12-29:
I can confirm.

After updating to da59cb1, I tried to watch a movie (V_MPEG4/ISO/AVC + A_AAC-2) and the screen froze.
I had a SSH session running, but even `systemctl shudown` and `reboot -f` failed.

dea7234 is working fine.

sairuk, 2021-01-07:
Any movement on this one, any additional info you lot need to help out?

pelwell, 2021-01-07:
Yesterday's firmware release (252ff0e943b4) should fix the audio problem.

sairuk, 2021-01-08:
Can confirm audio playback is working fine on 5467199574e4a6621ded42359dfd9aadfc4793ee

matzrm, 2021-01-08:
The commit 5467199574e4a6621ded42359dfd9aadfc4793ee fix my problem.

closed 2021-02-09 by matzrm