Hexxeh/rpi-firmware #251: Boot issue with latest 5.10.0 and 5.4.83 kernels

MrEngman, 2020-12-15:
Running rpi-update and BRANCH=next rpi-update and only the Pi4 keep working, Pi4 4GB, Pi4 8GB and Pi400.

All other Pi, Pi0, Pi0W, Pi1B, Pi2B, Pi3A fail to boot. 

Pi4 4GB and Pi400 with kernel 5.4.83 and Pi4 8GB with kernel 5.10.0 appear to be running OK.

Pi3A with kernel 5.10.0 and other Pi with kernel 5.4.83 all fail to boot. 

Master commit 453e49bdd8, BRANCH=next commit 6ac95f9714

MrEngman

mirak011, 2020-12-15:
According to this comment [https://github.com/raspberrypi/firmware/issues/1511#issuecomment-745192475](url)
it is a problem with GPU firmware. It can be solved so far by using previous _start*.elf, fixup*.dat, bootcode.bin_.

pelwell, 2020-12-15:
This is fixed in the new new firmware.

jjsuwa-sys3175, 2020-12-15:
LGTM (pi3b). thank you for your quick correspondence.

MrEngman, 2020-12-15:
Probem with my various Pi now fixed, Thank you

MrEngman

closed 2020-12-15 by popcornmix